supermerill/SuperSlicer #1328: Pressure adance is not disabled during TOOLCHANGE LOAD

MatthiasKohler, 2021-06-18:
### Version
2.3.56

### Operating system type + version
Ubuntu 18.04

### Printer + Firmware
Voron Switchwire with y-splitter, dual Bowden extruders, Klipper

### Behavior
A high value of pressure advance (>0.4) and the loading speed during TOOLCHANGE LOAD leads to more steps
being requested from the mcu than it can output leading to a mcu crash. SuperSlicer already disables pressure advance during  TOOLCHANGE UNLOAD, however it reenables it when the filament start gcode is called, which is before TOOLCHANGE LOAD. 

### Desired behavior
Keep pressure advance disabled during TOOLCHANGE LOAD. This could be achieved by having separate filament start gcodes  before and after loading.

Attached is part of a gcode file, where this happens.

[Test-Object-ESUN_ABS+-230_VoronSW.gcode.txt](https://github.com/supermerill/SuperSlicer/files/6679197/Test-Object-ESUN_ABS%2B-230_VoronSW.gcode.txt)

supermerill, 2021-06-18:
can you post also your project file? It's to see how your settings are made, to understand how the gcode is made.

MatthiasKohler, 2021-06-18:
[Test-Object.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6679512/Test-Object.3mf.txt)
You need to remove the .txt at the end. Github doesnt allow upload of files with extension .3mf

supermerill, 2021-06-22:
I don't understand.
You have set "SET_PRESSURE_ADVANCE" in your filament custom gcode, but that is a klipper gcode, right?
And you have set your printer firmware as "marlin", why?

for the issue, seems like the fix prusa made for their issue n°2770 is incomplete, as it rely on the filament start custom gcode, and sometimes it isn't called, ans so the PA is disabled until the next real toolchange.

So I think it's better to scrap all that and set the pa value in a setting field and let the slicer use it where it's needed.
That will need a comprehensive rewrite, so be patient.

MatthiasKohler, 2021-06-22:
Indeed the gcode flavor was set to Marlin. The problem persists if I change it to Klipper.

supermerill, 2021-06-22:
yes, and as said it's not easily fixable, as the way how the wipetower works have to be scrapped & remade.

MatthiasKohler, 2021-06-22:
What is wrong with the wipe tower? We have a script in Voronland which postprocesses a gcode file from Prusaslicer and simply disables PA before TOOLCHANGE LOAD and reenables it before TOOLCHANGE WIPE.

Perhaps there should be a command in klipper to tell it to not apply PA to specific G1 E....

supermerill, 2021-06-22:
post-process can work, but i won't create a post-process to fix a bug.
the wipe tower create gcode on its side, without much knowledge of the print.
then, it's copied into the output gcode.
Currently, the commit say that It will disable pa at a point and rely on the filament custom gcode to re-enable it, even if sometimes it's not called.
As your problem is after wieptower gcode, It doesn't have any way to block your customgcode (which shouldn't be erased) nor create another piece of gcode somewhere else.
It's not needed to add another 'PA disbling' command in the 'TOOLCHANGE UNLOAD', all you have to do is to not call the 'SET_PRESSURE_ADVANCE' in the filament custom gcode.

Currently, the software doesn't have your pa value as a setting, an so it's not managed by the slicer (or very poorly, as demonstrated here)