supermerill/SuperSlicer #1329: Milling post procesing: requests and bugs.

AmyTheCute, 2021-06-18:
**Describe the bug**
Bug: the gcode tries to switch my extruder to "extruder2" when it switches to the mill. it treats the mill like a third extruder while my printer has non. I fix this issue for now by using sublime text and removing all instances of that gcode

bug: the mill tries to mill way too far from the print  I tried increasing the extra xy size.

**To Reproduce**
enable milling post-processing.

**Additional context**

also I have a few feature requests if it's possible!
1) Multiple more gradual passes rather than a single pass.
2) only do the post processing every x layers? is there a way this would combine every few layers into one? say for 45 degree surfaces it would give them a smooth finish etc

supermerill, 2021-06-18:
Oh! a user! great!

> Bug: the gcode tries to switch my extruder to "extruder2" when it switches to the mill

Yes, that's the only way I could find to tell to the firmware "hey, switch from the extruder to the milling tool".
How you do the switch on your firmware?

>bug: the mill tries to mill way too far from the print I tried increasing the extra xy size.

when adding your milling tool to your printer, it has a mill diameter, maybe that's that which has a wrong value?

> Multiple more gradual passes rather than a single pass.

like a pass at 2mm dist and then a pass at 1mm dist? A setting like "max mm per cut" ?

> only do the post processing every x layers?

yes, maybe even a "max height" so when using variable height, it will do pass at the layer before it's too deep.

AmyTheCute, 2021-06-18:
Oh! a user! great!

> Oh! a user! great!
:P 

> Yes, that's the only way I could find to tell to the firmware "hey, switch from the extruder to the milling tool".
> How you do the switch on your firmware?

On my firmware I have a macro named (pick_mill) and it picks the mills and sets the output pin to ON. the slicer is having this in the gcode: Activate_extruder extruder=extruder2. this doesn't even pick a tool. it just sets the hotend/extruder pins to be the third one in which case my printer doesn't have a 3rd extruder, adding one would require a heater and extruder step pin etc which my printer doesn't have. I don't even have pins left on my board to use as a fake hotend.

>when adding your milling tool to your printer, it has a mill diameter, maybe that's that which has a wrong value?
on the slider? i set it to the correct number.

>like a pass at 2mm dist and then a pass at 1mm dist? A setting like "max mm per cut" ?
like instead of going all at once and shaving everything at once. do a pass that cuts 0.1mm of material and another than cuts another 0.1mm. or just pass on the same area twice. like those laser engravers do.

>yes, maybe even a "max height" so when using variable height, it will do pass at the layer before it's too deep.
yeah kinda like the fusion 360 ASMBL addon!

supermerill, 2021-06-19:
> On my firmware I have a macro named (pick_mill) and it picks the mills and sets the output pin to ON. the slicer is having this in the gcode: Activate_extruder extruder=extruder2. this doesn't even [...]

in the `custom toolchange gcode` write something like:
```
{if previous_extruder>=2 and next_extruder<2}unpick_mill{endif}
{if next_extruder<2}T{next_extruder}{else}pick_mill{endif}

```

> like instead of going all at once and shaving everything at once. do a pass that cuts 0.1mm of material and another than cuts another 0.1mm. or just pass on the same area twice. like those laser engravers do.

With `max_mm_per_cut=0.1`, it will cut 0.1 mm of thing, and then it will do another pass 0.1mm nearer (horizontally), and will continue until it's all cut. Is it ok?

AmyTheCute, 2021-06-19:
but why would you want to switch to an extruder anyways? why not get rid of that line compeletly and only add the "on tool change" gcode?

also I'm a moron. my endmill was 2mm.... I measured it wrong. my fault there. will run another test cube like that now!

>With max_mm_per_cut=0.1, it will cut 0.1 mm of thing, and then it will do another pass 0.1mm nearer (horizontally), and will continue until it's all cut. Is it ok? 

I'm not sure where that setting is. I can't find it!
but that'd be what I'm asking for exactly :D maybe later there can be even non plannar stuff added. or more smart cutting that only cuts every X layers like the fusion 360 cam.

supermerill, 2021-06-19:
> but why would you want to switch to an extruder anyways? why not get rid of that line compeletly and only add the "on tool change" gcode?

Ok, I get rid of the toolchange gcode. milling start & end gcode should be enough.

> I'm not sure where that setting is. I can't find it!

It's not made yet.

AmyTheCute, 2021-06-19:
> > but why would you want to switch to an extruder anyways? why not get rid of that line compeletly and only add the "on tool change" gcode?
> 
> Ok, I get rid of the toolchange gcode. milling start & end gcode should be enough.
> 
> > I'm not sure where that setting is. I can't find it!
> 
> It's not made yet.

Ooh thanks a lot for the help!

And your slicer is really awesome thanks for working on it!.

I should also add that I know NOTHING about CNC machines so I'm not a good source of info. But I do have a mill on my Doot Changer :p