supermerill/SuperSlicer #1330: new creality ender 3 bltouch conf will not add to printers

kaibsora, 2021-06-19:
### Version
_Version: 2.3.56.5_

### Operating system type + version
windows 11 + 10 latest

### Behavior
When trying to add the bltouch version of ender 3, and clicking finish from both brand new unzip of super, you get the prompt that the printer model ffff doesnt have a filament type selected, would you like to select default filament for the model. it doesnt matter what you select, it doesnt get added. 
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
No

supermerill, 2021-06-19:
reported here
https://github.com/supermerill/SuperSlicer/issues/896

seems the fix didn't make through

kaibsora, 2021-06-19:
Awesome! Seems I didn't go back so far. Thanks so much!

closed 2021-06-21 by supermerill