supermerill/SuperSlicer #1331: Performance Issue

Jaaannis, 2021-06-19:
**Describe the bug**
I have performance issues, when i load bigger .stl files, i can not use SS.
When i load just a 3DBenchy, in gcode preview, i have cpu load an all cores about 60% with bad performance(when i rotate the camera constant).
When i do this with PS on 3 core 15%.

**To Reproduce**
Load 3D Benchy, slice, rotate camera.

**Desktop (please complete the following information):**
 - OS: Win64 
 - Version 2.3.56.5
 - Ryzen 5 AMD 3600
 - 16GB RAM
 - AMD RX5700XT

Jaaannis, 2021-06-19:
i have do delete the opengl32.dll in SS directory, like here https://github.com/supermerill/SuperSlicer/issues/1163
performance after that is very well !

closed 2021-06-19 by Jaaannis

supermerill, 2021-06-21:
there is no opengl32.dll in SS directory in  2.3.56.5

aka13-404, 2021-07-24:
> there is no opengl32.dll in SS directory in 2.3.56.5

I just stumbled upon it, fixed it for me as well. The cause is propably that I am simply unpacking the zip of each new version, overwriting files. Guess the dll remained there from some older version. Not sure if/how it can be fixed automatically, maybe a small warning that such a thing could happen, when detecting low fps?

supermerill, 2021-07-25:
>  Guess the dll remained there from some older version. Not sure if/how it can be fixed automatically, maybe a small warning that such a thing could happen, when detecting low fps?

easier to detect if the file is here.