supermerill/SuperSlicer #1332: Add separate nozzle temperature setting for ironing

martinbudden, 2021-06-19:
When ironing considerable heat can be put into the print surface, since the hotend is moving slowly and repeatedly moves over the same area. This can cause undue melting and/or buckling of the ironed surface. This is particularly true if the nozzle temperature is set at the high end of the temperature range of the filament.

See the picture below for an example. There is considerable overmelting/buckling around the bottom two holes. I noticed this and manually turned down the hotend temperature by about 15 degrees while the print was still in progress. As you can see, the overmelting is much reduced over the rest of the print (in retrospect I should have turned down the temperature even more).

Having a separate ironing nozzle temperature setting would improve this situation.

[image]

supermerill, 2021-06-19:
Not really possible, if it's colder, it may not be able to extrude filament. And there is the delay to cool it then heat up again...

Why not just increasing the speed?

martinbudden, 2021-06-19:
> Not really possible, if it's colder, it may not be able to extrude filament

Well, that's up to the user to set a sensible temperature.

Often I set the hotend temperature at the high end - there are two main reasons for doing this: either for stronger prints, or for doing high volumetric speed printing (eg 0.6mm nozzle, 0.5mm layer height, 1.0mm extrusion width). So for PLA I might set a nozzle temperature of 230 or 240 degrees, depending on the PLA being used. However that is far too hot for ironing, where a temperature of 200 or so degrees needed. No problem extruding the filament for ironing at that temperature.

> And there is the delay to cool it then heat up again..

That is not a problem if you are ironing "Topmost surface only". And if you are ironing "All solid surfaces" or "All top surfaces" then you set a smaller temperature difference.

> Why not just increasing the speed?

Yes, I do that as well, but you still get melting/buckling problems.