supermerill/SuperSlicer #1333: Set "Use relative E distances" on by default

martinbudden, 2021-06-19:
There is no advantage in using absolute E distances, and doing so introduces the possibility of rounding error and/or the accumulated extrusion value used by the slice becoming out of step with the accumulated extrusion value used by the printer firmware.

supermerill, 2021-06-19:
When there is many small extrusions, "relative E distance" creates rounding error, and makes the printer to go out of step with the accumulation of these errors. Something that don't happens with absolute position. That's why it's on absolute by default.

martinbudden, 2021-06-19:
I hadn't considered that case. So in either case the printer can get out of step with the slicer, so there there is no clear-cut advantage to my suggestion.

Closing this issue.

martinbudden, 2021-06-20:
I'm reopening this because I had an idea about this.

Looking at the gcode I notices the extruder is reset using `G92 E0` every time there is a retraction.

It gave me the idea that the extruder could be reset every time there is a layer change. That way the slight disadvantage of using absolute E values is almost totally ameliorated, since the E values don't accumulate to large values.

supermerill, 2021-06-21:
Isn't it already the case?

martinbudden, 2021-06-22:
No, I don't think so. From looking at gcode it seems the extruder is reset every time there is a retraction, but it is not reset every layer change. For many prints there is at least one retraction per layer, so for those it makes no difference. But looking at some of my prints, some go for many layers without any retractions.

tikiss, 2021-09-11:
I have also noticed that even though I use absolute extrusion distance it will do a reset before and after retraction.
Which is causing (in my case) artifacts on the external perimeter, if I slicing the xyz cube with Cura (with absolute retraction distance) the cube surface is nice without artifacts.
Would be great if the absolute and relative extrusion distance works independently from the retraction (like in case of Cura).
If extrusion distance reset is needed before layout change, it is possible to add a costume G-code (before/after layer change), or?