supermerill/SuperSlicer #1338: When "Background processing" set focus changes to Gcode preview every time a setting is changed

martinbudden, 2021-06-20:
**Describe the bug**
If "Background processing" is set, then whenever a setting is changed, focus immediately jumps to the Gcode preview window.

**To Reproduce**
1. In the Configuration menu, Preferences dialog, on the General tab set "Background processing" to true.
2. Go to Print Settings and change any item.
3. The focus immediately changes to the "Gcode Preview" view.

**Project File**
Not applicable.

**Expected behavior**
When a Print setting is changed, the focus should remain on the Print settings page.

**Screenshots**
Not applicable.

**Desktop (please complete the following information):**
 OS: Windows
 Version: Windows 10 Home 21H1
 Superslicer Version: 2.3.56.5
 Build: SuperSlicer_2.3.56.5

neophyl, 2021-06-20:
Martin this is already there.  A new option was added to Preferences>gui>Switch from 3d view to preview when sliced.  If you untick that (its ticked to preserve the current behavior) then when changing settings it wont change from the screen you are on.

martinbudden, 2021-06-20:
Thanks, that does the trick.

Closing.

reloxx13, 2021-06-25:
this is sooooo f***ing anoying. should be auto set if background processing is on. senseless to have it on.

supermerill, 2021-06-25:
> this is sooooo f***ing anoying. switching should be off automated if background processing is on. senseless to have it on.

true

closed 2021-08-09 by supermerill