supermerill/SuperSlicer #1339: Feature request - Add setting for first layer travel speed

martinbudden, 2021-06-20:
**Is your feature request related to a problem? Please describe.**
Rapid travel movements on the first layer can cause problems with bed adhesion. For example, if the print is slightly stringy then a rapid movement can drag a just completed line segment. There is also the possibility that the nozzle can dislodge an overextruded (and hence proud) line segment as it passes over it.

**Describe the solution you'd like**
An option to set the travel speed for the first layer.

supermerill, 2021-06-21:
quick travel should help to "cut" the current extrusion.
Dis you make a test with low travel speed & high travel speed?
can i see the results (pictures)?

martinbudden, 2021-06-22:
> quick travel should help to "cut" the current extrusion.

This is true for PLA, but for PETG the current extrusion can be stretched out until it pulls the previously extruded part off the printbed.

> Did you make a test with low travel speed & high travel speed?

I haven't got any pictures of the problems and fixes, but I'll look into making some test cases and post the results. See also https://github.com/supermerill/SuperSlicer/issues/1216 - there are genuine problems getting a detailed first layer to stick and the more weapons we have in the armoury the better.

There are actually two issues here

1. Having an extra setting to help get the first layer to stick
2. General future proofing

Dealing with issue (2): Voron printers commonly set travel speeds of 250 to 300mm/s, and I have a printer where I am using those settings. This is fine for most layers, but having the printhead moving at that speed on the first layer is just asking for trouble. Currently the only way to avoid that is to lower the travel speed for all layers.

Also not that there is already a setting for Acceleration for the first layer, so it is odd to be able to control acceleration on the first layer but not speed.

supermerill, 2021-06-23:
> Also note that there is already a setting for Acceleration for the first layer, so it is odd to be able to control acceleration on the first layer but not speed.

there is a modifier for speed for the first layer. Just not for travel specifically.

martinbudden, 2021-06-23:
Do you want me to submit a PR?

supermerill, 2021-06-23:
If already has a working one, why not.

martinbudden, 2021-07-01:
I don't already have a working PR, but it should not be too difficult.

I don't mind if the PR is deleted later if you implement it in some future change, so whether I do the work for the PR depends on when you plan the future change. If you plan to do the work next week or next month, then there is little point me submitting a PR. But if you don't plan to do it for 6 months or more, then I can do the PR.

SamSpiri, 2022-05-27:
Have you got any workaround?

seancmonahan, 2022-05-27:
At the moment, I've just been using a hacky Python script that lets me do find/replace in a single layer. Travel moves emitted by SuperSlicer are always '^G1 [^E]+$' (G1 with no 'E' anywhere in the line (after chopping any comment component at the end of the line).

Regardless, a possible, slightly less hacky solution might be (untested yet) adding `M203 X50 Y50` to your start gcode (limiting your X and Y axis feed rates to 50mm/s each), with a layer gcode macro that resets them to whatever your printer max is usually if the layer number is 2. (On mobile right now, so I can't type up full syntax. I'll edit this later with a working fix.)

SamSpiri, 2022-05-29:
In order to get everything automated, were you able to run it as a post-processing script?

seancmonahan, 2022-05-30:
[limit_first_layer_travel_speed.py ](https://gist.github.com/seancmonahan/05f92505162a7081b1539686ad33f535).

Add `/path/to/limit_first_layer_travel_speed.py [desired first layer travel speed in mm/s]` to Print Settings -> Output options -> Post-processing scripts. I've only tested it with SuperSlicer 2.4.58.3 (nightly from 2022-05-18) on Linux. It seems to work both with and without verbose g-code. It might _not_ work if for some reason you have your travel speed set the same as any first layer extrusion speeds, because SuperSlicer omits redundant `F` parameters.

Fr00t, 2023-01-12:
> > Also note that there is already a setting for Acceleration for the first layer, so it is odd to be able to control acceleration on the first layer but not speed.
> 
> there is a modifier for speed for the first layer. Just not for travel specifically.

I was a bit stomped by this before i found this thread. To me its weird to have a modifier called "First layer acceleration" that doesn't affect travel moves. 

The script doesn't provide changes in acceleration as far as I can see, and I'm struggling with fast acceleration moves on the first layer that pulls up small perimeters.

I tried looking at the relevant code but this is miles above my league :D

Henk72, 2023-01-18:
Would love to see this option in SuperSlicer