supermerill/SuperSlicer #1340: Bridges not neing bridges

Tinchus2009, 2021-06-20:
**Describe the bug**
Im seeing these bridges being created where actually doesnt seem to be necesary. As you can see in the pictures, a normal perimeter is created, interrupted, a small piece of bridge created and then a normal perimeter is created again.

**To Reproduce**
I have included the project file

**>> Project File <<**

**Expected behavior**
MAy be there is a probblem in the detection of the angle of a wall? 
**Screenshots**

[image]
[DUCT_strange_bridges.zip](https://github.com/supermerill/SuperSlicer/files/6682525/DUCT_strange_bridges.zip)

**Desktop (please complete the following information):**
 - OS:win10 
 - Version Version 2.3.56.1

**Additional context**
Add any other context about the problem here.

danielfmo, 2021-06-20:
Check the Overhangs section under Perimeters & Shell. 
From there I get that the percentage of overlap with the bellow layer is also used to determine if it is a bridge/overhang

neophyl, 2021-06-20:
iirc that version has an issue with bridge detection that has been fixed since (at least it looks ok in 2.3.56.5).  https://github.com/supermerill/SuperSlicer/issues/1186

Tinchus2009, 2021-06-21:
IT is not an overlap setting issue, because in the image you can see that normal perimeter is created, then a very small portion of bridge is created and then again it  goes back to normal perimeter again. Also, the angle of that wall is below below 45 degrees in order to need bridges on that border?. Version 2.3.56.5 shows the same behavior

supermerill, 2021-06-22:
imo it's due to a too low overhang threshold.
There is almost no bridge on the nightly, because i removed a unneeded 'simplify geometry'.
the slope is almost exactly on your overhang threshold.

> Also, the angle of that wall is below below 45 degrees in order to need bridges on that border?

'overhang threshold' doesn't work with° but with mm of overhangs (width of the unsupported section)

Tinchus2009, 2021-06-25:
ok... but is is very strange to se the generation of a normal path, then only 1 mm or less of a bridge and then normal path, and then a bridge again ans so on like in the ima ge above. Ans the overhang threshold there was 50%. That means 50% of the extrusio width, wich clearly s not the case on the blue areas. That areas, the angle of that wall is bigger thatn 45 degrees and layer height is 0.2, so there should not be detected as bridges IMHO

supermerill, 2021-06-25:
>  is very strange to se the generation of a normal path, then only 1 mm or less of a bridge and then 

yes, have to add a min length (see an other issue)

> wich clearly s not the case on the blue areas.

it's exactly at 50%, I even took pictures and count pixel to prove it:
[image]
If you really want me to loose time on this, i can extract the polygons that are created on it.

closed 2021-07-20 by Tinchus2009