supermerill/SuperSlicer #1341: [2.3.56.5] Width / Spacing Issue

CCS86, 2021-06-20:
Not sure what is going on here, since the new width / spacing has generally been working well for me.

I have a cylindrical shell which is 1.6mm wide:

[image]

With the following settings, I expect 4 extrusion lines and no gap fill:

[image]

But, here is the result:

[image]

[WidthIssue.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6682935/WidthIssue.3mf.txt)

CCS86, 2021-06-20:
It actually looks like this is a front-end / GUI issue. 

I had worked down to spacing values of 0.36mm and still the path hadn't changed. Then, I set the width values to 0.4mm (which bumped the calculated spacing up to 0.37xx, and suddenly I had 4 lines, plus a little gap fill. I then set the spacing values all back to 0.4mm and the path was correct.

It seems like (at least when setting spacing values) something gets "stuck" in the GUI and doesn't pass the correct values to the engine. I'm curious what my imported project looks like to someone else.

poernahi, 2021-06-21:
Seems to work as expected on my computer (2.3.56.5 windows)

However, I notice my slicer produce different path. Do you see the same result if you restart the slicer, reload the file and re-slice?

[image]

supermerill, 2021-06-21:
Modifying a spacing doesn't reset the slicing data like widths do. have to fix that.

closed 2021-07-10 by supermerill