supermerill/SuperSlicer #1342: Clarify Infill Settings

CCS86, 2021-06-20:
2.3.56.5 / Win10 x64

I thought I understand the infill labels/settings, but apparently I am confused.

At the top you have the label "sparse"; which in this context, to me, means "less than 100%. You then have an input for density, and a pattern selection. The most logical thing to me would be that if density was set between 1% - 99%, the engine would use the pattern chosen here, for all infill which was not on a "top"/"bottom" layer. If density was set to 100%, it would no longer be "sparse" and the pattern (for non-top/bottom) would be chosen from the drop-down just below labeled "solid".

That just doesn't seem to be that way it works. Here I have density set to 100%, 0 top layers, 0 bottom layers. I would assume that all my infill was the same "solid infill" and would also be "filled":

[image]

Instead, I get mostly "internal infill", which is not "filled"; and a tiny amount of "solid infill", only on top layers.

[image]

[image]

It sure looks like I am mostly getting the "normal" rectilinear pattern chosen in the "sparse" field (no "filled" option available there).

Can someone help me understand this better? Thanks!

[InfillType.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6683195/InfillType.3mf.txt)

neophyl, 2021-06-21:
You are getting normal 'sparse' infill at 100% there as that is how your project is configured.  You do have a bit of solid infill shown in pink though.

The rectilinear(filled) infill is not an option for the normal sparse infill so of course you get the small gaps as they aren't filled in with gapfill like they would be if Rectilinear(filled) was available.  This is normal as internally infill is generally not needed to be solid.

If you wanted actual solid infill then there are a couple of options.  
You could increase the number of top and bottom layers till there are no internal layers left.  That's a common way to do it.  Probably the most common.

The other option is the solid infill threshold setting.  I mentioned you had some solid infill in your current project.  That's because you have the solid infill value set to 70mm2 .  Any area less than that is being filled in with solid infill.  Which as it is set to Rectilinear (filled) is using gapfill on those little areas.  Increase the threshold and it will do the rest with solid infill too.  

The types of infill are for different uses/areas.  For sparse infill not all infill patterns are allowed to be set to 100%.  Rectilinear is as its basically the same as the common top/bottom infill.  Try setting some of the others to 100%, iirc it tops out at 99%.   The (filled) options are available in SuSi but aren't part of normal PS so you get the slight gaps normally as it cant fit a normal extrusion in there.  For example concentric normally has a small gap in the centre.  Concentric(filled) has the extra gapfill.  However you will note that none of the (filled) versions are available for sparse.  PS doesnt apply gapfill to any infill, just normal extrusion.  It does seem to use the normal extrusion differently though.

CCS86, 2021-06-21:
Thanks for the insight @neophyl 

I do a lot of functional prints, so small gaps in the infill:perimeter can cause failure points for me.  

Definitely not the most intuitive part of the configuration. I had been using the inflated top/bottom layer value as a workaround. But, it's good to have that solid infill threshold as an option as well.

closed 2021-09-06 by supermerill