supermerill/SuperSlicer #1346: Features for continuous fiber printing

ShirdelY, 2021-06-21:
I am currently working on a DIY continuous fiber extrusion system, and have the mechanics figured out. Currently, I am coming up with a software solution.

because the fiber cutter isn't directly at the nozzle tip, but about 60mm above the nozzle, you would need to cut the fiber 60mm before a retraction. Ideally, I would be able to insert a gcode (servo gcode for example) x distance before a retraction, so that the remaining fiber can be dragged out right up until the retraction, where the fiber would be cut and the print head is free to move elsewhere.

Another side effect of the distance between the cutter and nozzle is that any fiber that is printed has a minimum distance of +60mm. So having 2 printheads is necessary, so if a perimeter or line of infill is longer than the minimum fiber length, it will use the fiber print head, and if the extrusion line is shorter, it will use a non-fiber extruder.

supermerill, 2021-06-21:
* how to tell the software where do you want your fiber?
  * in all perimeters / all infill ?
  * via a modifier?
* then you can reject all lines that are shorter than your cutter
  * all lines that are kept have to use the extruder X instead of the extruder Y ?
* then, when writing gcode, you have to insert the cut gcode 60mm before the end.

that's your plan?

ShirdelY, 2021-06-21:
for the line selector, I was thinking of selecting different line types, for example, internal perimeters and infill, or just perimeters. And for the inserted gcode, it would only be necessary for lines made by the fiber extruder. I'm not sure how this would be implemented, but an example of this would be if you had a straight line for example 100mm long. then the slicer would need to split it into a 40mm and 60mm section, and insert something like "M300 S90" or something similar.

ShirdelY, 2021-06-21:
[image]

ShirdelY, 2021-06-21:
oops just realized the point are on the wrong lines. the red dot should be on the internal perimeter next to the external perimeter