supermerill/SuperSlicer #1348: Deretraction after color change (M600) causing blob on print

rcclausen, 2021-06-22:
**Describe the bug**
The gcode generated from the slicer with a color change still has a deretraction after the M600 command. However the nozzle is already filled with plastic so it creates a small blob on the print.

This is a bug that was also reported and apparently fixed for PrusaSlicer, however it doesn't appear to be in the current release:
https://github.com/prusa3d/PrusaSlicer/issues/6362#issuecomment-820468245

**To Reproduce**
Steps to reproduce the behavior:
1. Add any object to the slicer
2. Slice it with any profile
3. Add a color change somewhere on the model
4. Export the gcode
5. Find the M600 command, it should be followed 2-3 lines by a Deretraction

**Expected behavior**
I would expect that the deretraction is omitted or that a retraction ahead of time is added since the nozzle is already primed with plastic

**GCode**
`;COLOR_CHANGE,T0`
`M600`
`; INIT printing object f_tail_1_1x.stl id:0 copy 0`
`; printing object f_tail_1_1x.stl id:0 copy 0`
`G1 X201.026 Y145.391`
`G1 Z2.000000`
`G1 E0.80000 F1800.000`    <<<<====== This line should not be here
`;TYPE:External perimeter`

**Desktop:**
 - OS: Windows
 - Version 2.3.56.5

supermerill, 2021-06-22:
I think that commit was made after ps 2.3.1
I may ba able to cherry-pick it.

rcclausen, 2021-06-22:
That makes sense. It did seem close to release cutoffs. For the time being, I am can just delete that line when I do color changes. Thanks!

closed 2021-07-10 by supermerill