supermerill/SuperSlicer #1349: [FR] Avoid Crossing Perimeters: Inset Distance

CCS86, 2021-06-22:
When using the "avoid crossing perimeters" function, the engine hugs whichever outer wall it would have crossed. It appears to land directly on the most outer wall's printed path:

[image]

The problem with this, is that retracing that path at high speed creates visible artifacts in that wall surface. For a lot of the same reasons mentioned in the "poyholes" blog post, you get filament deposited outside of the outer wall.

I think this can be solved completely with an "inset distance" parameter added to the feature. This would define a minimum distance for the travel path, in relation to that outer wall. Maybe it would be even simpler to specify a wall number, and the travel move could move from the outermost wall, in some number of walls.

qwewer0, 2021-06-23:
I like the wall number solution more, as that would adjusts to the line width.

CCS86, 2021-06-23:
It could be like many of the settings where you can express as a number in mm, or as a percentage of extrusion width.

supermerill, 2021-11-07:
It's a bit too complex to go deeper than the second perimeter without creating travelling problems. And nobody would want the previous version, right? So for now, no inset distance.

CCS86, 2021-11-07:
I'll give it some testing, but I'm pretty sure this will work well.

closed 2021-11-07 by CCS86