supermerill/SuperSlicer #1351: Gap Fill Missing Large Gaps

CCS86, 2021-06-23:
2.3.56.5 / Win10 x64

Even with Minimum gap fill area set to 0, it is missing some relatively large gaps:

[image]

[gapfillmiss.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6698876/gapfillmiss.3mf.txt)

neophyl, 2021-06-23:
When I try and open your projectct with 2.3.56.5 on Win10 (same version as you) I get the following error ConfigBase::set_deserialize() failed for parameter "fill_pattern", value ""  So it looks like one of the fill settings is missing when saved.  I'm wondering if that could be contributing to the missing gapFILL.

supermerill, 2021-06-23:
oh, that saving bug.
I finally succeed to replicate it and fix it not so long ago.

closed 2021-07-10 by supermerill