supermerill/SuperSlicer #1354: Shift in the numbering of layers in gcode preview for a bridge.

erathos77, 2021-06-24:
macOs 10.13, SS 2.3.56.5

Object to print as example: a bridge between 2 vertical walls (layers 1-124) spaced 20mm apart. PETG without support, layer 0.2mm height. Default settings. Output verbose g-code.

By examining the gcode built by Superslicer in a text editor, I see that the overhang perimeter and the bridge infill occur at layer 124. Then, at layer 125, a 45 ° infill begin. This does not match superslicer's gcode preview, which shows the overhangs perimeters and bridge infill built at layer 125. 

Context: I don't like to print any support with PETG. So, when possible, I prefer to pause the print and lay out a pre-made support covered with blue tape which will receive the bridge avoiding its deformation.

Looking only  the gcode preview, I would mistakenly assume that I should place the pause instruction at the end of layer 124 and before the start of 125,  so, one layer too late.

I suggest correcting this misleading mismatch unless a simple setting of the parameters would have escaped my attention. Thanks.

[test_bridge.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6708907/test_bridge.3mf.zip)

supermerill, 2021-06-24:
Does the numbering start at 0 in your gcode file?

erathos77, 2021-06-24:
OK you are right, it starts at 0 in the gcode file while at 1 in the gcode preview, thus explaining a fake upshift. But is there a good reason to generate a mismatch between the gcode  sequence and that of the gcode preview ? This is misleading if you don't read the gcode file with attention.
I'm guessing you'll answer that you just have to know it once and for all ! Now, I will never make again this mistake, but what for other users ?

supermerill, 2021-06-24:
>  But is there a good reason to generate a mismatch between the gcode sequence and that of the gcode preview ? 

Prusa decides to begin at 1 in the gcode preview and I didn't care changing it to begin at 0.
99.99% of the time, people don't read gcode file anyway, and personally, I rely mostly on Z height, not on layer count.

I just read a gcde file: I don't even know where you found the layer number.
that's what i use to find my layer:
```
;LAYER_CHANGE
;Z:0.5
```

erathos77, 2021-06-24:
Yes this is the layer numbering. I understand your arguments and finally agree with you. Thanks

closed 2021-06-25 by supermerill