supermerill/SuperSlicer #1359: The problem with semicircular walls 2.3.5хх

MrDavidFantasy, 2021-06-26:
**Describe the bug**
When printing semicircular or circular walls on the penultimate line in front of the outer wall, the head moves in jerks, which leads to the creation of overflows at the stops.

The problem exists in all slicers of versions 2.3.5хх
There is no such problem in version 2.2.54.

**Screenshots**
[https://disk.yandex.ru/i/6MCfKNRx2A5sLA](url)

The screenshot shows how surface defects are present in the right shape. This test was done on three printers with different kinematics.

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [e.g. 2.3.55.2 - 2.3.56.6]

supermerill, 2021-06-26:
your project?

MrDavidFantasy, 2021-06-26:
> it prints a bit, stops, then reprints, then stops?

During circular printing, he makes it with small delays in certain places. Randomly. He just lingered for a split second, and drives on. Where it lingered, a little plastic leaks.

> Can I have your project file, where it doesn't print well?

[Circle.zip](https://github.com/supermerill/SuperSlicer/files/6720053/Circle.zip)

supermerill, 2021-06-26:
the project file is the file created when you do "file -> save as"

MrDavidFantasy, 2021-06-26:
> the project file is the file created when you do "file -> save as"

Sorry :)
[Circle.zip](https://github.com/supermerill/SuperSlicer/files/6720148/Circle.zip)

Not quite sure where to change the resolution?

supermerill, 2021-06-26:
yeah, you should decrease resolution and/or up your min length (printer tab) by a good amount (something like 0.35 mm instead of 0.035)

test that and report plz.

MrDavidFantasy, 2021-06-26:
> test that and report plz.

Thanks friend, the min length did not help, but changing the resolution from 0 to 0.01 corrected the situation. Thank You!

MrDavidFantasy, 2021-08-05:
Hello! I had to reopen the thread on my problem. The fact is that I continued my experiments. And I managed to find out that at small radii (10mm) the problem appears again. I tried increasing the resolution - this partially solves the problem, but some freezes (and leaked plastic) are still present. Then I noticed that the finished file in version 2.3.56.7 takes 4.2 megabytes, while in version 2.2.54 the same files take 711 kilobytes with the same settings. There is something else in the new version that greatly overloads the code. Help me please.

supermerill, 2021-08-06:
on my computer
your project is 3.1 mo in 2.2.54
your project is 5mo in 2.3.56 with resolution to 0.0125
your project is 3.6mo in 2.3.56 with resolution to 0.0125 and minimum extrusion length to 0.35

and you didn't post your new project file(s). How can i help you?

MrDavidFantasy, 2021-08-06:
the old project is already in the past. Now I was printing small radius parts and it was with them that the problem came back. Here's a new project. Thanks
[38-1.zip](https://github.com/supermerill/SuperSlicer/files/6944818/38-1.zip)

supermerill, 2021-08-06:
I have 3mo for 2.3.56 and 1.8mo for 2.2.54
Seems like when loading the project, the 2.3.54 didn't read correctly the 'thin_perimeter' setting (overlapping external perimeter), and checked it.
If you also check it on the 2.3.56 version, it will change the slicing from thin_wall to perimeter. And as teh periemter have a constant flow, it doesn't need that many commands.

CCS86, 2021-08-08:
> the old project is already in the past. Now I was printing small radius parts and it was with them that the problem came back. Here's a new project.
> [38-1.zip](https://github.com/supermerill/SuperSlicer/files/6944818/38-1.zip)
> 
> It is this project that I have very different in size:
> 2,9 mb in 2.3.56
> 0,7 mb in 2.2.54
> Why is there such a difference in size?
> It seems to me that this is exactly the problem of freezing.
> Thanks

Your min extrusion length is 0.04. That is way too small. Merill already told you this.

Set min extrusion length=0.3, and resolution=0 and try again.

MrDavidFantasy, 2021-08-08:
Thanks everyone! Still, the problem is that the min extrusion length.

closed 2021-08-08 by MrDavidFantasy