supermerill/SuperSlicer #1360: Supress switching to slicer preview when changing settings

scheini79, 2021-06-26:
### Version
2.3.56.6

### Operating system type + version
Win10 Pro

### Behavior
* When in settings tabs (Print, Filament, Printer), changing a setting switches automatically to sliced preview tab.
* _Steps needed to reproduce the problem_ none
* _Expected Results_ stay in settings tabs until i want to choose sliced preview
[Glorious Bruticus-Sango.zip](https://github.com/supermerill/SuperSlicer/files/6719905/Glorious.Bruticus-Sango.zip)

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2021-06-26:
duplicate of #1338 ?

scheini79, 2021-06-26:
> 
> 
> duplicate of #1338 ?

Yep, haven't seen it in the settings. Solved. Thanks.

closed 2021-06-26 by supermerill