supermerill/SuperSlicer #1361: [FR] Only one perimeter on Bottom/First layer

randellhodges, 2021-06-26:
I know I can use the  "Height Range Modifier" feature to adjust the number of perimeters for the first layer, but if you have plate full of parts, you have to use the merge feature first, so you can do it once. Otherwise, you'll spend the day clicking each item.

If you merge all the parts, would that not impact where the z-seam is place when you use the align feature? If it does, then you wouldn't want to use the merge workaround.

supermerill, 2021-06-27:
yes

duplicate of #412

randellhodges, 2021-06-27:
Well, the workaround, using a "height range modifier" is indeed a duplicate of #412. Sorry about that. I didn't clearly state the feature request. In addition to #412, a checkbox for a single perimeter on the first layer might be useful as it would be part of the print settings vs the object settings.