supermerill/SuperSlicer #1369: First Layer XY compensation adding a unnecessary perimeter line

HateFindingNames, 2021-06-29:
**Describe the bug**
If _Bridge flow_ threshold is set it interferes with _First Layer XY compensation_ aka "elephant foot compensation" in a way that an extra perimeter line gets added on the second layer with a strange line height. This behavior eliminates the original purpose of _First Layer XY compensation_.

**To Reproduce**
Steps to reproduce the behavior:
1. Set a negative value for First Layer XY compensation while having set a value for _Bridge Flow_ threshold about 50%.
2. Slice the model.
3. Get introduced to an extra perimeter line on the second layer.

**>> Project File <<**
[Project File](https://github.com/supermerill/SuperSlicer/files/6734080/Shape-Box.zip)

**Expected behavior**
Setting any value for XY compensation should shrink/expand the first layer by the given value keeping all other parameters untouched.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win10 64bit
 - Version 2.3.56.6

**Additional context**
-/-

supermerill, 2021-06-29:
why open two issues?
I can't spot the difference between these two.

HateFindingNames, 2021-06-30:
My apologies. Indeed the issue seems to originate from _Bridge flow_ setting as it affects all layers. I will change the title accordingly.

[image]
The gif above illustrates the effect of having _Bridge flow_ set to values of 20% and 100% (mind the legend).

[image]
Here you can see the effect on a 45° slope. What I would expect is having the _Bridge flow_ set to 100% will generate a line with layer height as set under _Slicing_ and a line width as set under _Width & flow -> external perimeter_. Actual behavior is a line generated way too thick (.4 mm height in this case).

supermerill, 2021-06-30:
if you don't want the brdge flow but the fan speed for overhangs, deactivate the perimeters -> overhangs-> threshold -> bridge flow (set to 0)