supermerill/SuperSlicer #1371: Question: Benefit of "Wipe While Retracting"?

CCS86, 2021-06-29:
At first glance, this sounds like a very useful feature. Instead of staying parked on an external perimeter during the retract move, allow the nozzle to move away at the same time.

But, when I look at the paths with and without this feature enabled, I'm struggling to see how it is helping.

No wipe:
[image]

With wipe:
[image]

I would expect that the wipe move would be a very short inwards move, normal to the external wall. What we see is a very long move, retracing the external perimeter, and also causing the retract move to be performed significantly slower, because it gets dragged out over a large XY distance and many individual segments (~0.1mm long).

`;WIPE_START
G0 X122.928 Y105.662 F10800
G1 F8640
G1 X122.917 Y105.683 E0.01917
G1 X122.816 Y105.680 E0.02343
G1 X122.715 Y105.672 E0.02343
G1 X122.614 Y105.659 E0.02343
G1 X122.514 Y105.640 E0.02343
G1 X122.416 Y105.617 E0.02343
G1 X122.318 Y105.588 E0.02343
G1 X122.271 Y105.571 E0.01172
G1 X122.223 Y105.554 E0.01172
G1 X122.129 Y105.515 E0.02343
G1 X122.037 Y105.471 E0.02343
G1 X121.948 Y105.423 E0.02343
G1 X121.862 Y105.370 E0.02343
G1 X121.778 Y105.312 E0.02343
G1 X121.697 Y105.251 E0.02343
G1 X121.620 Y105.185 E0.02343
G1 X121.547 Y105.115 E0.02343
G1 X121.477 Y105.041 E0.02343
G1 X121.411 Y104.964 E0.02343
G1 X121.349 Y104.884 E0.02343
G1 X121.292 Y104.800 E0.02343
G1 X121.239 Y104.713 E0.02343
G1 X121.190 Y104.624 E0.02343
G1 X121.146 Y104.533 E0.02343
G1 X121.108 Y104.439 E0.02343
G1 X121.091 Y104.391 E0.01172
G1 X121.074 Y104.343 E0.01172
G1 X121.045 Y104.246 E0.02343
G1 X121.021 Y104.147 E0.02343
G1 X121.003 Y104.047 E0.02343
G1 X120.989 Y103.947 E0.02343
G1 X120.982 Y103.846 E0.02343
G1 X120.979 Y103.744 E0.02343
G1 X120.982 Y103.643 E0.02343
G1 X120.986 Y103.581 E0.01437
;WIPE_END`

Can someone explain the benefit here?

supermerill, 2021-06-29:
You activate the wipe, so it will wipe the nozzle (ie moving the nozzle over the current loop) to clean it to ensure there is no/less stringing.

CCS86, 2021-06-29:
Gotcha. Seems to be working as intended then.

Maybe it leaves room for a similar feature, like I was expecting, where the nozzle dives away from the external wall before initiating the retract move.

It looks like this current wipe feature suffers from the same issue I mentioned here: https://github.com/supermerill/SuperSlicer/issues/1210

closed 2021-09-06 by supermerill