supermerill/SuperSlicer #1373: Colour change bar shows wrong colours

hagan66, 2021-06-30:
I often do colour change at z-height. The bar does not show right colours. It does at first then it gets messed up.

I just set a sequence of colour changes and then it gets messed up. Print is still ok.

[image]

supermerill, 2021-06-30:
can you post your project file?

hagan66, 2021-06-30:
[moai-new.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6740590/moai-new.3mf.zip)

While recreating this file, I discovered its the "no sparse layers" option for the wipe tower that mess up the colours. :-)

One workaround is to do whatever colour change settings before turning on the "no sparse layers" option.

supermerill, 2021-07-02:
That's because fo the "no sparse layers (EXPERIMENTAL)"
I won't fix it as prusa will obviously fix their experimental feature themselves in the short term.
You can use the 'sliced preview' that will display the bar correctly. To get to it you have some options:
 * use the tab layout, but it can be buggy on macos
 * change a gcode-only setting like 'printer notes' and the displayed view will silently change from gcode preview to sliced preview.