supermerill/SuperSlicer #1374: Big Sur app icon style

LazaroFilm, 2021-06-30:
**Is your feature request related to a problem? Please describe.**
MacOs Big Sur has a guidelines to make all icons with a similar style. SuperSlicer's icon doesn't currently follow those guidelines

**Describe the solution you'd like**
A redesigned icon that follows [those guidelines](https://developer.apple.com/design/human-interface-guidelines/macos/icons-and-images/app-icon/)

LazaroFilm, 2021-07-01:
I created an icon for it and you can download it [HERE
[image]](https://mega.nz/file/4hR0XLrI#mPwECHIY4SxzqOwcqurQi_b0vBJmtrm4m2p4LQHN8jI)
I took [this icon made by Jacopo Bassan](https://dribbble.com/shots/14651990-PrusaSlicer-macOS-Big-Sur-Icon) and changed the hue from orange to blue.

liftbag, 2021-07-01:
> I created an icon for it and you can download it

The link is about the PrusaSlicer icon? It's orange.

supermerill, 2021-07-02:
Can you do the same with the slic3r colors? (you can have the correspondence in the preference-> gui color fields, on the tooltips).

LazaroFilm, 2021-07-03:
Oops. I have corrected the link in my previous post. 
Also here is a [Mega Link with a folder containing all three icons](https://mega.nz/folder/x1gz1aKL#64sl3yN_KVakn_Ivvq73kQ). The original by Jacopo Bassan, the SuperSlicer and the Slic3r. 
Slic3r was a bit tougher, I had to play with more than just Hue to get the yellow and it's not perfect. 
[image]

supermerill, 2021-07-06:
Is it working on the latest nightly?

liftbag, 2021-07-07:
> Is it working on the latest nightly?

Almost. The dock icon changes appearance when the SuperSlicer is running.

https://user-images.githubusercontent.com/14921654/124815945-2ed16600-df68-11eb-9905-53a871990a2b.mp4

LazaroFilm, 2021-07-12:
@supermerill Thank you so much for making this happen so quickly! It's so gratifying to see something you contributed to come to the release!

closed 2021-08-09 by supermerill