supermerill/SuperSlicer #1376: Print host Repetier Server not working when the host is not called Repetier-Server

Prutsium, 2021-07-01:
Version 2.3.56.6
When selecting Repetier as Print Host Upload it wants to see the Repetier to be called Repetier-Server.
This works for standard installs but often Repetier is Branded and have their own name like CR3D-Server due that the Repetier connection wont work as it doesnt pass the version check.

Suggested to have or a naming box or a checkbox to skip the check.

supermerill, 2021-07-02:
Can you put here the error messages? I can't spot anywhere a 'Repetier-Server' check.

CageFox, 2021-07-02:
I always using repetier server as host and my host name not even close to "Repetier-Server" and all works well. My host have name "P" and printers named like "P" and "SP" ;)

Prutsium, 2021-07-02:
It's not related to the host name ....
It's the branding name that can be changed for by resellers.
[image]

CageFox, 2021-07-02:
Is there a way to check this name in existing installation?

supermerill, 2021-07-02:
@Prutsium 
Can you put here the error messages? I can't spot anywhere a 'Repetier-Server' check.

Prutsium, 2021-07-08:
See added screenshots hope they would help
[image]
[image]
[image]

The printers are detected but not possible to add

closed 2021-08-09 by supermerill