supermerill/SuperSlicer #1382: Confirmation dialogue for deleing print profiles

Niper8, 2021-07-01:
**Is your feature request related to a problem? Please describe.**
Just clicked the delete profile button thinking it was the discard changes button (good thing I make backups :) )

It would be nice if it confirms that you want to delete a profile.

**Describe the solution you'd like**
Ask the user if they are sure they want to delete.

**Describe how it would work**
Are you sure you want to delete it? yes/no

supermerill, 2021-07-02:
Can't reproduce, I have a confirm dialog.

[image]

Can you explain in detail how you manage to avoid it?

Niper8, 2021-07-02:
Well, nor can I ,😂. I think I must have been on autopilot

closed 2021-07-02 by Niper8