supermerill/SuperSlicer #1383: Bridge infill erroneously sliced as gap infill for wide extrusion widths

martinbudden, 2021-07-01:
**Describe the bug**
Quite an interesting bug - when using wide extrusion widths (1mm in the example below), `bridge infill` can be erroneously sliced as `gap infill`.

This bug also exists in PrusaSlicer.

A bit of background which might explain why this bug hasn't surfaced before. I'm probably a bit unconventional in my default print settings. I print mainly functional parts and my default print settings are `0.6 mm nozzle`, `0.5 mm layer height`, `1.0 mm extrusion width` with a standard E3 v6 hotend (not volcano) and max volumetric speed set to 14 mm<sup>3</sup>/s. It seems most people use much smaller layer heights and extrusion widths which is probably why this bug hasn't surfaced before.

**To Reproduce**
Load the attached file and slice it. The problem is observed for all bridges, and is most easily observed at the first bridging layer, that is at layer 6.

**>> Project File <<**

Here's the zipped project file
[6_bridging_test.zip](https://github.com/supermerill/SuperSlicer/files/6750635/6_bridging_test.zip)

And here's the zipped config settings
[config.zip](https://github.com/supermerill/SuperSlicer/files/6750636/config.zip)

**Expected behavior**
For the first layer of the bridges, the perimeters should be `overhang perimeter`s and the infill should be `bridge infill`.

**Screenshots**
Here's a screenshot showing the first bridging layer. Note that the two outside features of the bridge are correctly `overhang perimeters`, whereas the middle of the bridge is `gap infill`. Note that this is printed with perimeters set to 2.

[image]

Here's a screenshot showing the speed, note that the `gap infill` of the bridge is printed slowly (and as a result droops right down onto the printbed).

[image]

Funnily enough, if perimeters is set to 1, the correct behaviour ensues:

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Home 21H1
 - Superslicer Version 2.3.56.5

supermerill, 2021-07-02:
As the space left between the periemter isn't wide enough for a third one, the slic3r decide to not use infill insied as it's too narrow for a good infill and decide to gapfill it.
The problem is that only infill area are checked for bridging....
Very problematic but also hard to fix.

martinbudden, 2021-07-02:
Why is it hard to check gapfill for bridging?

supermerill, 2021-07-02:
because it's not an area tagged for infill and there is already extrusion inside it.
Have to pre-check for bridging, then clip them with the gapfill to not extrude it, then create a special infill list for special briding thingys, then create a special step to create this special bridge.

martinbudden, 2021-07-04:
I notice you have a project to rewrite the gcode writer. Perhaps this should be added to that project?

supermerill, 2021-07-04:
it's well before that

SirStigII, 2021-07-11:
would it be possible to have the slicer check the bridging width and optimize the extrusion width to eliminate gaps like this? I've also noticed this happening, and really the only current fix is to play with the width settings to try and come up with better overall width settings, but on more complex prints that won't really work everywhere.

supermerill, 2021-07-11:
I have to rework the bridging process. I don't have the free time for that in the near future.