supermerill/SuperSlicer #1385: name not update after generate calibration object

tjengbudi, 2021-07-03:
**Describe the bug**
the name is not update after used to generate calibration object

**To Reproduce**
Steps to reproduce the behavior:
1. geberate calibration object. i used bridge calibration
2. export gcode and then save or cancel
3. delete generated object. add another stl manually
4. export gcode. the name is not the first object in the list

just minor bug, after used calibration tool

supermerill, 2021-07-03:
There is a weak form of "project".
The .3mf is the project file.
The name in top bar is the project name.

To create a new project, use file -> new project
or delete all -> new project

To name/rename a project, use save/as

The gcode file use the name of the project, or the first object if the project has not a name yet.

tjengbudi, 2021-07-04:
> enerating a calibration creates a calibration project. I

looks like i do not know how superslicer gave a name. so it must not a bug. thanks

closed 2021-07-04 by tjengbudi