supermerill/SuperSlicer #1386: Character display error

sziga, 2021-07-03:
I started translating SuperSlicer into Hungarian.
Some long accented characters like "ő" and "ű" appear incorrectly in the small windows.
In the main window, in settings they appear fine, in search, Variable layer hight,cut, Gcode preview top left information area window etc the "ő" "ű" characters are replaced by "?".
Version: SuperSlicer_2.3.56.6_win64_210626
Displayed in the settings: 
[image]
Search windows:
[image]
Variable layer hight windows:
[image]
Cut windows:
[image]
Gcode preview top left information area window:
[image]
The .po file includes the line "Content-Type: text/plain; charset=UTF-8\n":
[image]

supermerill, 2021-07-03:
give me your po, so i can test it.

sziga, 2021-07-03:
The translation is not yet complete and there are still a lot of mistranslations, but I'll send you the po file.
[hu.zip](https://github.com/supermerill/SuperSlicer/files/6758949/hu.zip)

supermerill, 2021-07-04:
Found it:
That's because I have to add the hungarian ranges for glyphs creation.
[image]
Will be fixed in next version
I'll add a post here when a nigthly will be built with it

sziga, 2021-07-04:
Okay. Thank you!

closed 2021-07-04 by sziga

AntoszHUN, 2022-04-03:
It's funny because there is no Hungarian language in it.

supermerill, 2022-04-03:
there is in the 2.4

AntoszHUN, 2022-04-03:
> 

> there is in the 2.4

However, it is not included in the new version.