supermerill/SuperSlicer #1388: Seam painting - Stop automatically processing as soon as you take your finger off the button!

skellatore, 2021-07-04:
**Is your feature request related to a problem? Please describe.**
When using the seam painting feature, the second you stop clicking/painting the system takes this as an opportunity to start processing automatically. This is quite infuriating as it automatically swaps to the sliced preview instantly and starts crunching.. you have to either wait for it to finish or get lucky with clicking the cancel button then going back to 3D view.

**Describe the solution you'd like**
An additional button like "Confirm"/"Process"/"OK" in the Seam Painting dialogue box would be more than ample to fix this.

skellatore, 2021-07-04:
I should note I _DONT_ have background processing enabled.

supermerill, 2021-07-04:
note that you can disable the "auto-switch" in the preferences

skellatore, 2021-07-05:
Thank you Super Merill - this is why they call you Super!

closed 2021-07-05 by skellatore

royal32, 2021-07-06:
Is this intended behavior? I've also been experiencing automatic slicing after every seam-painting click and It's incredibly frustrating. Disabling auto switch does prevent it from happening, but also (obviously) disables the view switching to the sliced view when the job is sliced normally, which is just as frustrating.
IMO Seam painting should be treated like every other object manipulation eg. you just click slice when you're done manipulating the object. It wouldn't slice every time you drag a part around and let go.

Thanks,
Seth

supermerill, 2021-07-06:
> Is this intended behavior?

no