supermerill/SuperSlicer #1389: [Linux] Opening an STL and having it open SuperSlicer instead of importing, breaks it forever. Crashes when viewing layer view

ChaosBlades, 2021-07-04:
**Describe the bug**
If you Double-Click an STL and open the file in SuperSlicer/PrusaSlicer, slice it, then switch to layer view it will crash. Then even if you delete the .config folder then relaunch SuperSlicer it will always crash when switching to layer view after slicing. I can not get it to work again after producing this bug. If you never have an STL open SuperSlicer/PrusaSlicer and always import them from within the application everything works fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Open an any STL and have the file open SuperSlicer/PrusaSlicer
2. Slice the file
3. Open layer view
4. The application will now crash and no matter how you import an STL from this point the application will always crash when opening layer view. Even if you delete the .config folder.

**Desktop (please complete the following information):**
 - OS: Linux Pop!_OS 21.04
 - Version The last 5ish versions I tested. Also tried SuperSlicer-ubuntu_18.04-2.3.56.6.AppImage and SuperSlicer-ubuntu_18.04-2.3.56.6b.AppImage In addition the latest version of PrusaSlicer also has this issue.

ChaosBlades, 2021-07-04:
[Crocodile_clip_v1.2.zip](https://github.com/supermerill/SuperSlicer/files/6759510/Crocodile_clip_v1.2.zip)

I imported this STL from within SuperSlicer and it crashes every time when opening layer view. I don't even need to have the STL open SuperSlicer. This is after completely re-installing the OS from scratch.

ChaosBlades, 2021-07-04:
Cura works without any issues. Guess I am going back to Cura until this is resolved. 

Let me know what more you need from me.

supermerill, 2021-07-05:
Can't reproduce.
Did you try the 2.3.56.6b version?

ChaosBlades, 2021-07-05:
> Version The last 5ish versions I tested. Also tried SuperSlicer-ubuntu_18.04-2.3.56.6.AppImage and **SuperSlicer-ubuntu_18.04-2.3.56.6b.AppImage I**n addition the latest version of PrusaSlicer also has this issue.

Did you test on Pop!_OS 21.04 or Ubuntu?
https://pop.system76.com/

I just re-tested on my Desktop and Laptop. It is no longer breaking forever now. Only the Crocodile clip is crashing it now.  Every time. This is a completely fresh install there is no config changes that I am making. Not sure what was going on 2 days ago but now only the Crocodile Clip is giving me crashes.
1. Open SuperSlicer (has no .config / new install)
2. Click Cancel on wizard
3. Add Crocodile Clip STL
4. Click "Slice now"
5. Application Crashes

I am noticing it is not crashing when slicing, it is crashing when processing all the calculations for the color coding filters.

supermerill, 2021-07-05:
> Did you test on Pop!_OS 21.04 or Ubuntu?

on ubuntu 20.04.2

ChaosBlades, 2021-07-05:
I'll re-test with Ubuntu 20.04 and 21.04. This test print STL is also causing the crash.
[AIl_In_One_3D_printer_test_micro.zip](https://github.com/supermerill/SuperSlicer/files/6766123/AIl_In_One_3D_printer_test_micro.zip)

supermerill, 2021-07-05:
works for me
[image]

ChaosBlades, 2021-07-05:
Did more testing. It seems to be an issue specifically with 21.04 on AMD hardware.

Laptop with Intel / Nvidia
Ubuntu 20.04 = Works
Ubuntu 21.04 = Works

Laptop with AMD CPU/GPU
Ubuntu 20.04 = Works
Ubuntu 21.04 = Crash

My Desktop that this is also crashing on is also 21.04 with AMD CPU/GPU. I also have one more system that is AMD CPU w/ Nvidia GPU if testing on that needs to happen let me know.

Do you have something to reproduce this on? Or have some way for me to collect the data you need to figure out what is going on here? If it is an issue with Ubuntu I would like to submit an issue upstream so it can be fixed.

supermerill, 2021-07-05:
I have an amd cpu and nvidia graphic, but I don't think the issue can occur on a vm.
Probably an issue with the opengl driver. Something he doesn't like.

ChaosBlades, 2021-07-05:
I tested everything on bare metal with a USB SSD.

Do you want me to submit an issue on the PrusaSlicer github or do you have an idea on how to work this issue?

ChaosBlades, 2021-07-05:
I made an issue with PrusaSlicer anyway since I can reproduce this with it as well.
https://github.com/prusa3d/PrusaSlicer/issues/6677

supermerill, 2021-07-06:
> Do you want me to submit an issue on the PrusaSlicer github

Only if you can reproduce the bug with prusaslicer.
Also, if possible, test it with prusaslicer 2.3.1 and 2.3.2, as it may be already fixed in the latest rc.

senseisimple, 2021-07-09:
I have submitted this issue on PrusaSlicer's githup repo (https://github.com/prusa3d/PrusaSlicer/issues/6677), posting it here for cross reference;

I'm having the same issue on Manjaro Arch, gnome 3/gtk3/qt5 , AMD Vega10 gpu.

Same trigger (as soon as done slicing right when feature colors are applied) and symptom (crash) as noted in both github issues for  PrusaSlic3r and SuperSlicer starting at version 2.3.0 to current with existing configs and new configs.

prusaslicer 2.20 (and superslicer 2.2.53) both work flawlessly.

base system specs:
```
Linux 5.11.4-1-rt11-MANJARO #1 SMP Fri May 21 00:10:17 CEST 2021 x86_64 GNU/Linux

#qtdiag
Qt 5.15.2 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 11.1.0) on "xcb"
Platforms available : 
  gtk2,gtk3,qt5ct,snap,flatpak,xdgdesktopportal
Available physical devices:
  API version 1.2.178, vendor 0x1002, device 0x6861, Radeon Pro WX 9100, type 2, driver version 2.0.188
Vulkan:
  Min Texture Size: 1
  Max Texture Size: 16384
  Max Color Attachments: 8
  Frames in Flight: 2
  Uniform Buffer Alignment: 16
  Supported MSAA sample counts: 1,2,4,8

#lspci (graphics)
01:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge
02:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XT [Radeon PRO WX 9100]
03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 HDMI Audio [Radeon Vega 56/64]

#lsmod (gpu)
amdgpu               6729728  98
gpu_sched              45056  1 amdgpu
i2c_algo_bit           16384  1 amdgpu
drm_ttm_helper         16384  1 amdgpu
ttm                    86016  2 amdgpu,drm_ttm_helper
drm_kms_helper        282624  1 amdgpu
drm                   569344  22 gpu_sched,drm_kms_helper,amdgpu,drm_ttm_helper,ttm

#glxinfo (opengl/vulkan)
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 16384 MB
    Total available memory: 32752 MB
    Currently available dedicated video memory: 13884 MB
OpenGL vendor string: AMD
OpenGL renderer string: Radeon Pro WX 9100 (VEGA10, DRM 3.40.0, 5.11.4-1-rt11-MANJARO, LLVM 12.0.0)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.1.2
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: no-error
OpenGL core profile profile mask: core profile

#clinfo (opencl)
$ sudo clinfo
Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 21.1.2
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 1
  Device Name                                     Radeon Pro WX 9100 (VEGA10, DRM 3.40.0, 5.11.4-1-rt11-MANJARO, LLVM 12.0.0)
  Device Vendor                                   AMD
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.1 Mesa 21.1.2
  Device Numeric Version                          0x401000 (1.1.0)
[... 70 lines cut ...]
02:49:01 AM: Debug: window wxTreeCtrl(0x55ff68e1a800, ) lost focus even though it didn't have it
02:49:01 AM: Debug: window wxTreeCtrl(0x55ff68e1a800, ) lost focus even though it didn't have it

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.814: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.814: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.902: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.902: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
Segmentation fault (core dumped)

```

dmesg output (relevant info):
```
#prusaslicer:
[1917053.404093] traps: slic3r_main[1809560] general protection fault ip:7f37ac588bdf sp:7ffd444fbcb0 error:0 in radeonsi_dri.so[7f37ac3ad000+fda000]
[1917053.404127] audit: type=1701 audit(1625814789.091:2419): auid=1000 uid=1000 gid=1000 ses=3 pid=1809560 comm="slic3r_main" exe="/tmp/.mount_PrusaS7GI8WE/usr/bin/bin/prusa-slicer" sig=11 res=1

#superslicer:
[1918672.551957] superslice:gl0[1819292]: segfault at 3560 ip 00007fbf1a7b0239 sp 00007fbeedff81a0 error 4 in radeonsi_dri.so[7fbf1a707000+fda000]
[1918672.551968] Code: 00 4c 89 fe eb cb 0f 1f 00 4c 89 fe eb 83 0f 1f 00 48 89 df e8 08 83 ff ff e9 c1 fd ff ff 0f 1f 00 48 85 d2 0f 84 45 ff ff ff <48> 8b 8a c8 00 00 00 48 85 c9 0f 84 2c ff ff ff 48 3b aa d0 00 00
[1918672.551996] audit: type=1701 audit(1625816408.219:2454): auid=1000 uid=1000 gid=1000 ses=3 pid=1819227 comm="superslice:gl0" exe="/usr/share/SuperSlicer/superslicer" sig=11 res=1
```

ChaosBlades, 2021-07-12:
Can confirm PrusaSlicer and SuperSlicer 2.2 both work without issue. Something in 2.3 is causing this issue.

senseisimple, 2021-07-13:
[broken-stl-example.zip](https://github.com/prusa3d/PrusaSlicer/files/6806175/broken-stl-example.zip)

Here is an example of an object that makes the sliced preview crash. The crash happens within 1mm (actually a 0.2mm layer) of a working cut. I have included .stl (SuperSlicer 2.3+ creates non-manifold cut when cutting sections), .obj (repaired manifold, same issues), .3mf, .gcode (sliced with SuperSlicer 2.2.53).

That is from a complete object that makes the slicer preview crash, I have isolated the 2.3*mm section that makes a reproducible crash, 
Mashroom (20)-25.49mm-top-2.30mm-OK (stl, obj) - slices/previews OK
Mashroom (20)-25.49mm-top-2.31mm-broken (stl, obj) - crashes on slice/preview

senseisimple, 2021-07-13:
It seems to have been narrowed down to the perimeter width and gapfill settings as detailed in https://github.com/supermerill/SuperSlicer/issues/1313#issuecomment-878803443 and further confirmed in the next comment.

Excerpts:

> I narrowed this down to the MedialAxis call when generating gapfill extrusions. This particular layer has some thin walls where only two perimeters and a big fat gapfill would fit.

> - Width and Flow: anything other than Default: 0.4mm (int/ext Perimeter 0) (0.2mm layer height) crashes
> - Infill: Top and/or Bottom: "Monotonic ( filled),", "Concentric (Filled) (any perimeter width) crashes
> - **Infill: Top and Bottom: Rectilinear (no fill) @ Width and Flow: Default: 0.4mm (int/ext Perimeter 0) (0.2mm layer height) NO CRASH but**

supermerill, 2021-07-13:
can you test with the latest nightly?

ChaosBlades, 2021-07-13:
Can you link to the appimage, I am not finding it.

senseisimple, 2021-07-14:
for nightly i don't think there is an appimage, has to be compiled as far as i know. You may find success running your current 2.3+ install from command line, setting an environment variable to disable hardware OpenGL rendering:

```
export LIBGL_ALWAYS_SOFTWARE=true
superslicer
```
or
```
LIBGL_ALWAYS_SOFTWARE=true superslicer
```
I'm not sure at the moment if the appimage provides a `superslicer` exec in $PATH. I'm pretty sure you can set the environment variable (export ....) and launch the *.appimage directly from the command line.

ChaosBlades, 2021-07-14:
I don't know how to compile this from source nor do I have the time to learn. If someone has an appimage for me to test I can do that. Until then I am more than happy to continue to use Cura rather than using a work around to use SuperSlicer/PrusaSlicer.

supermerill, 2021-07-16:
> Can you link to the appimage, I am not finding it.

https://github.com/supermerill/SuperSlicer/actions/runs/1020922835
https://github.com/supermerill/SuperSlicer/suites/3211044497/artifacts/74282099

ChaosBlades, 2021-07-16:
No change, it is still crashing just like before. I did delete all configuration files before testing.

ChaosBlades, 2021-09-18:
I recently got an update to Mesa 21.2.1 which appears to resolve this issue.

closed 2021-09-18 by ChaosBlades

senseisimple, 2021-10-06:
Not solved for Arch with mesa 21.2.2-1, SuperSlicer 2.3.57.0 though the original "broken" stl files now slice and render ok and we can handle slightly more complex models, large slices still cause a seg fault on loading the slice preview at the point of coloring/labeling the feature types (or other views with a legend).