supermerill/SuperSlicer #1391: Infill not generated on multiple layers

herrtschaefer, 2021-07-04:
The infill is not generated in the middle of the object, it stops and on other layers the infill is generated again. I also did run into the issue with the last release and some other infill patterns cubic and adaptive cubic and now gyroid. I did not try other infill patterns yet. Sometimes they do show up in the sliced tab but not the g code  tab. I also viewed it in the duet web g code viewer and no infill lines where shown in the middle of the print. 

I attached a zip file containing the stl,3mf,slicer info and config bundle.
[issue_files_missing_infill.zip](https://github.com/supermerill/SuperSlicer/files/6760105/issue_files_missing_infill.zip)
 
The image shows the model in the duet gcode viewer, on the bottom the infill is generated (red) and then there is a large gap with multiple layers without infill (there should have been a lot of red and blue lines) and then at the top most layer shown here the infill pattern begins again.
[image]

Can a post processing script like ArcWelder influence the view in the slicer? I thought that the script only runs when exporting the g code.

supermerill, 2021-07-05:
That's your toolchange_gcode that creates this preview problem.
The slicer don't understand the 
````
G60 S3               ;Save current position
G1 Z5 R3            ;Lift Z
T[next_extruder] ;Tool change
G1 X0 Y0 R3       ;Move over last position
G1 Z0 R3            ;Move Down
```

supermerill, 2021-07-06:
lightly built

closed 2021-08-09 by supermerill