supermerill/SuperSlicer #1396: Feature request - add support for twisted polyholes

martinbudden, 2021-07-05:
A while back I worked with [NopHead](https://github.com/nophead) to add twisted polyholes to the OpenSCAD library [NopSCADlib](https://github.com/nophead/NopSCADlib), see https://github.com/nophead/NopSCADlib/issues/106

I noticed you have added polyhole support to SuperSlicer, I think it would be relatively straightforward to extend that to support twisted polyholes.

A twisted polyhole is a polyhole where the polygon rotates with each layer of the print. They have a number of advantages over standard polyholes

1. If they are near the edge of an object, are stronger, since the infill is in different places for each layer.
2. I think they are generally stronger, although I haven't done any calibrated tests to prove this
3. Cosmetic - they look like round holes rather than polygons.

Including twisted polyholes in the slicer would make them available to people who don't use OpenSCAD and NopSCADlib.

I've used them extensively and have found them effective. See https://github.com/martinbudden/BabyCube for a project that makes extensive use of them.

Here's a part that uses twisted polyholes:

[image]

Here's a view of that part from above, showing how the holes look circular (this is also the case for the part, once printed).

[image]

And here is a series of screenshots showing the first 4 layers of the print. Look in particular at the polyholes near the edges of the part and how they interact with the perimeters and the gap infill.

[image]

[image]

[image]

[image]

supermerill, 2021-07-05:
what's best?
* rotating for X° per layer
* alternating 0 - half turn (60° for a triangle, 45° for a square ...)
* both
* other variant

martinbudden, 2021-07-05:
I use "other variant"...

> rotating for X° per layer

This isn't good enough, since you may choose a value that causes layers of a particular polygon to be coincident

> alternating 0 - half turn (60° for a triangle, 45° for a square ...)

This only gives 2 different layers, so is sub-optimal.

What I have been using is "alternating every n layers". I've been experimenting with `n=3` and `n=5` and I've settled on `n=5`. So that means 120/5° for triangle, 90/5° for square etc.

Because twisted polyholes add many facets to the model, they really slow things down in OpenSCAD, for that reason I also have use a cut-off, I don't use twisted polyholes when the hole diameter is 4mm or more, since the benefit is less for larger holes. However since speed won't be so much an issue for the slicer, I'd set the cutoff higher - maybe use twisted polyholes up to and including 5mm. (Or maybe no cutoff at all, I'm not sure).