supermerill/SuperSlicer #1398: Weird behavior with bridging

thomaspj1017, 2021-07-07:
### Version
SuperSlicer Version 2.3.56.6

### Operating system type + version
Windows 10

### Behavior
For parts with a one layer thick wall beneath an unsupported hole, rather than do straight bridging across, it creates a bridge circle which defeats the purpose of the sacrificial layer.

Below is a screen shot with the one on the left showing the issue and the one on the right showing how I want it/it should perform for unsupported vertical holes.

[image]

supermerill, 2021-07-10:
Untick "only one perimeter on top" or at least at this layer, with a 'height range modifier'

Because it's a one-layer surface, it's tagged as "top", and so the algorithm shift other perimeters.
You're the second one to report that.
Maybe I can to check for bridging before adding these perimeters.

Zeblote, 2021-12-23:
I just encountered this strange behavior aswell.

Prusa Slicer:
[image]

Super Slicer:
[image]

Unchecking "only one perimeter" "on top surfaces" fixed it. However, this is not a top surface!
[image]

supermerill, 2021-12-23:
@Zeblote
you forgot to add your project, so i can't test, but I guess that the difference here is from the 'anchor solid infill'  bridge margin that is set to 3mm in prusaslicer (and also the default one).

Zeblote, 2021-12-24:
Good point. The file I was trying to slice is here: https://main.eva-3d.page/hotends/rapido/

And the project: [bridge_infill_bug.zip](https://github.com/supermerill/SuperSlicer/files/7771994/bridge_infill_bug.zip)

supermerill, 2022-01-31:
> However, this is not a top surface!

Not a top surface but there's nothing on top of it. 
[image]

supermerill, 2022-01-31:
for the "overhang" detection, it's deactivated in the linked project. You just ahve to reactivate it.
for the problematic bridging area, it's a duplicate of issue #1494

OldCurmudgeon3DP, 2023-04-08:
I've been running into the same issue as the OP when trying to avoid using supports for vertical holes in parts. I'll try unchecking the 1 perimeter option. I found that it does it using both the built-in primitives and if I have a 1 layer thick "bridge" built in from the modeling software (Fusion).