supermerill/SuperSlicer #1399: Massive overextrusion on small diameter objects

hagan66, 2021-07-08:
Hi! 

Massive overextrusion on small diameter perimeters. Example shows spikes, but also small holes behave the same way.

Everything is fine and dandy, then I got this. :-)

Prusa mk2.5s
0.5mm nozzle
PLA
Superslicer Mac version 2.3.56.6

[image]
[Massive over extrusion problem.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6784182/Massive.over.extrusion.problem.3mf.zip)

supermerill, 2021-07-10:
Weird, as prusa output almost the same amount of plastic.
The obvious thing is the different amount of gapfill outputted. So i'll change it a bit to avoid the star-shape gapfill.
If you deactivate gapfill, will it be better?

hagan66, 2021-07-10:
I actually suspected the gapfill, and turned it off but no difference. Hmm.

supermerill, 2021-07-10:
If we can't identify the root cause, I'll have to print them and iterate many time, will be a much more slower progress.

hagan66, 2021-07-10:
I will be happy to do some more testing. 👍🏻

closed 2021-07-10 by supermerill

hagan66, 2021-07-10:
I think this is something. Both cylinders are 4mm diameter. Left one is a standard cylinder with lots of segments/polygons. Heavy overextrusion. It also stutters and prints very slowly. The right one has only 12 segments and prints ok.
[image]

I hope this helps.

supermerill, 2021-07-10:
Too much resolution setting, or/and too low min extrusion length.
I really have to improve the "min extrusion length" to take into account the speed.

CCS86, 2021-07-12:
I think the only "issue" with min. extrusion length, is that the behavior changed dramatically in one of the recent releases.

Setting up a simple test in 2.2.54 vs 2.3.56.6, adding a 20x20x20 cylinder, and slicing with the same layer height, width, min ext length, 1 wall, 0 top/bottom. 

[image]

[image]

[image]

The newer version generates 3 times more print segments. 

[2.2.54.gcode.txt](https://github.com/supermerill/SuperSlicer/files/6802443/2.2.54.gcode.txt)
[2.3.56.6.gcode.txt](https://github.com/supermerill/SuperSlicer/files/6802444/2.3.56.6.gcode.txt)

CCS86, 2021-07-12:
Changing the cylinder to 3x3x100 shows the differences even more, as it leans on the min ext length harder to avoid path deviation:

[image]

[image]

Some of the gcode analyser values for the 2.3.56.6 code don't make sense to me: the amount of travel moves, total distance moved and the total Z height. 

But dividing the total distance traveled by the number of print moves gives:

2.2.54:  0.278 mm
2.3.56.6: 0.048 mm

This is for a 0.035 mm min ext length setting. So, the newer version is much more true to the value. 

You just need to get people to update their profiles. In my testing, I find that [min extrusion length] = 0.2 mm gives a comparable number of print segments to 0.035 mm in the earlier versions.

hagan66, 2021-07-12:
I just upped the minimum extrusion length value as much as possible and while still getting accaptable details I landed at 0.3 mm. I also compared this to Prusaslicers preview and it looks about the same. Works fine now.

:-)

/Robert