supermerill/SuperSlicer #1400: Underextrusion on long distances?

nicedevil007, 2021-07-09:
**Describe the bug**
[image]

The picture says everything, the left object is printed with SuperSlicer and the right one with IdeaMaker.
I get those underextrusions on every print that has such long lines to print in front of a corner section.

**To Reproduce**
These are my settings in SuperSlicer for the extruder and I use relative Extrusion:
[image]

[config.zip](https://github.com/supermerill/SuperSlicer/files/6788906/config.zip)
**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.56.6

JohanAR, 2021-07-09:
I think it will be easier to debug if you save and attach the whole project (a .3mf file) which includes both the model you're slicing and settings

I saw that your config.ini that you have "post_process = ArcWelder.exe".. Did you try printing it without ArcWelder?

nicedevil007, 2021-07-09:
Ah that is a good point, can try again without! I will report back what the results are! If they are bad I can send the 3mf file :)

nicedevil007, 2021-07-10:
seems to fixed without acrwelder :) ty

closed 2021-07-10 by nicedevil007