supermerill/SuperSlicer #1402: Strange overextrusions surrounding features

AndrewEllis93, 2021-07-09:
**Describe the bug**
**SS is producing strange overextruded perimeters surrounding certain features:**
[image]
[image]
[image]

**This issue does not appear to be present when printing with Prusa Slicer:**
[image]

I and a few other folks in the Voron Discord have been experiencing this issue and have yet to come up with a reliable solution.

**To Reproduce**
See attached .3MF file for settings used. I have been unable to narrow this issue down to a particular setting, and I have left many of the advanced features at defaults.
[SS-Issue.zip](https://github.com/supermerill/SuperSlicer/files/6794002/SS-Issue.zip)

**Desktop (please complete the following information):**
Windows 10 Pro
SuperSlicer version 2.3.56.6

supermerill, 2021-07-10:
If you can reliably reproduce the artifact, can you try by removing "only one perimeter on top" ?

AndrewEllis93, 2021-07-10:
> If you can reliably reproduce the artifact, can you try by removing "only one perimeter on top" ?

Just tried - same exact lines appear.

Someone mentioned that they were able to reduce these lines by reducing top layer flow - however that's more of a clue than a solution, as I can't lower my top layer flow without creating gaps.

I asked them to provide their photos/results on this GitHub issue.

supermerill, 2021-07-10:
It's weird. Like the extruder stop, plastic overflowing from it. Then it moves again but the pressure isn't here yet.
Maybe too many small segments that overload the firmware/serial port ...

AndrewEllis93, 2021-07-10:
My MCU loads appear to be okay - do you suggest I try lowering slicing resolution?

supermerill, 2021-07-11:
You can, but i don't think it's the issue, as in your picture the problem appears in the middle of a long straight line.

Tinchus2009, 2021-07-20:
Can this be related to my obseved problem? because it is what IM seeing in some print but much worst: #1427

supermerill, 2021-07-20:
That may be. You can try to increase the xy decimals (to 5-6) to see if the firmware react better.

AndrewEllis93, 2021-10-04:
While still not perfect, the best solution I found was to simply reduce perimeter accelerations to a low value. Closing

sarchar, 2022-04-25:
[image]

I'm having this issue and I've been fighting it for a couple months now.  I have a test print where I set to top solid layers to 0, which actually removes all internal solid layers.  The result is that the object with internal solid infill doesn't exhibit the artifact, while the one with solid infill does:

[image]

I think this issue should be reopened..?

sarchar, 2022-04-26:
I printed another test stopping the print right after the artifact layer so that I can inspect the infill. Here's what I have.

One object is printed with "0 top layers" -- it disables all the internal solid infills as well (that's weird TBH, but it does).

Gcode preview (in this image the right object has 0 top layers):
[image]

You can clearly see the artifact on the object (sorry, they're reversed from the ones in the gcode preview):
[image]

The solid infill looks insanely over-extruded:

[image]

It seems the artifact follows the direction the nozzle is flowing. It shows up only at the end of the extrusion. The artifact is also visible here for that reason, I just can't get a good photo of it:
[image]

Some print settings: 0.2 layer height, 100% solid infill width, 140% sparse. Sparse infill 200mm/s, solid 100mm/s, top solid 60mm/s.

sarchar, 2022-04-27:
> Because the perimeter isn't supported by a solid when there isn't (at least one) solid top layer(s), then the pressure in the nozzle can't be built up, and so can't be released (at the bad place).
> Why your printer is doing that... maybe a bit of over-extrusion (can be enforced by low pa / high accel/low jerk) is creating that, releasing the pressure at the first position where an open side allows for easier overflow.

I'm fairly certain I printed this test at various PA values and both lowered and raised accels to no effect.  I'll retest both of those variables soon.  What do you mean by "low jerk"?

supermerill, 2022-04-28:
> What do you mean by "low jerk"?

With no/low jerk, it decelerate to a stall before accelerating in the other direction.

IMO, from the look of the picture, it seems you're over-extruding a bit, and that may be the root issue.

MoBeltrao, 2022-06-20:
> > What do you mean by "low jerk"?
> 
> With no/low jerk, it decelerate to a stall before accelerating in the other direction.
> 
> IMO, from the look of the picture, it seems you're over-extruding a bit, and that may be the root issue.

I had exact the same problem and this post save a lot of job trying to figure out.
I was over-extruding as you said....thank you, nice tip.

AndrewEllis93, 2022-11-09:
@sarchar @MoBeltrao @Tinchus2009 

I spent a bit of time digging into [issue 2665](https://github.com/supermerill/SuperSlicer/issues/2665).
Lo and behold, my solution for that issue also turned out to have a significant improvement for this issue too. (for me, at least).
- Documented [here](https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/articles/troubleshooting/small_infill_areas_overextruded.md#not-connected-top-infill-superslicer).
- My comment on the other issue [here](https://github.com/supermerill/SuperSlicer/issues/2665#issuecomment-1306455603).

I'm curious if you guys are able to replicate this.

To summarize:
- Top solid: set to "not connected"
- Solid: also set to "not connected"
- Infill/perimeter overlap (encroachment): Set to 25%

The three settings below also bring some more improvement when combined with the above.
However they come with some downsides described in the linked doc above (either potential top layer scarring or excessive z-hopping on the top layers, take your pick)
- Minimum travel after retraction: 0
- Z-hop: Disabled on top layer
- eOnly retract when crossing perimeters: Enabled