supermerill/SuperSlicer #1405: Flow::mm3_per_mm() produced negative flow  Error when Support enabled

xTremePower, 2021-07-10:
**Describe the bug**
Hello,

I Use the MX version for my Newstar Z9V5, the colors works perfectly, but I have a problem when I activate Support,
If I activate Supports the slicer  say me "Flow::mm3_per_mm() produced negative flow. Did you set some extrusion width too small?"

I just found how to correct this, but don't understand why it does this,
In printer settings, In used Extruders, (here 1) If there was no 0 in Layer Height limits (here 3mm) the bug happens, if I set 0, it works again

Thanks

**Screenshots**
[image]
[image]

neophyl, 2021-07-10:
A 3mm thick layer from a 0.4mm nozzle isn’t going to work. Why is it set that high ?

supermerill, 2021-07-10:
It's made to create supports as tall as possible to reduce the time to print them. So it uses the max layer height. If it's ~ higher than the nozzle diameter, the flow math will bug out.

I should add a warning popup on this field