supermerill/SuperSlicer #1408: 2.3.56.7 no longer sets the correct tool when printing (regression from 2.3.56.5)

johnmeacham, 2021-07-13:
[hollow_cube.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6808772/hollow_cube.3mf.zip)
**Describe the bug**
When  changing an extruder to be non default it does not emit the appropriate toolchange code to change the tool. It does set the bed temperature from the chosen extruder so ends up printing with the wrong temperature on the wrong tool.

**To Reproduce**
Steps to reproduce the behavior:

Load an object, click on the right under extruder, choose a tool other than default.
Export the gcode and the Tn commands and the custom toolchange code are not included. However temperature commands for the tool it should have used are still in there.

**Expected behavior**
When a tool other than default is selected, it should emit the appropriate tool command to choose that tool and include the user custom toolchange gcode.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version [e.g. 2.3.55.5],  2.3.56.7
 
**Additional context**
This worked properly in 2.3.56.5 

Diffing the two, other than floating point formatting, omitting the toolchange code appears to be the only difference between the .5 and .7 versions.

supermerill, 2021-07-16:
maybe because of #1377 ?

johnmeacham, 2021-07-25:
Hmm.. that might be a workaround, will have to test,  but the fact it prints with the wrong temp by default makes me think it is a more serious issue. 

I think logic that would lead to consistent behavior for the M600 and toolchanger cases would be to always and only emit tool change code if anything was set to not be 'default' in the extruder config for the object.  counting the number of changes and trying to guess can lead to very suprising behavior such as if you delete an object off your build plate and it suddenly changes the tool with which the other objects print.

M4rkusHartmair, 2021-08-06:
I´m printing on a E3D Tool-Changer with one Tool. 
Since 2.3.56.7 SS doesn´t  set any Tool G-Code at the start, most likely intentional with #1377 .
This printer has to home without a tool, so a initial Tool-Call is absolutely necessary. ATM my printer attempts to start printing without a Tool. This behavior almost destroyed my Z-Probe on the Toolhead. The firmware doesn't apply the defined T0 XYZ offset and attempts to "print" on a height of 0.1mm below the micro-switch trigger-point. 

Adding a "T[current_extruder]" like in "Tool change G-code" apparently doesn't work, there is still no "T0" Call at the start. 

Is there a workaround except for manually editing every single File by hand?

supermerill, 2021-08-06:
@johnmeacham I looked again, and if it doesn't write the 'T', it's because you wrote a 'T' in your custom start gcode, and so it assume you already set it.

@M4rkusHartmair
I'm sorry. There is plenty of very different multi-extruder setup, it's impossible to keep track of all. Maybe you can share the problematic project? Maybe I can add a failsafe for a specific configuration?

M4rkusHartmair, 2021-08-07:
I´m using a modified version of the SuperSlicer-config-bundle profile distributed by E3D here: 
https://github.com/e3donline/Slicer-Profiles/tree/master/SuperSlicer

Ran a Test and it actually sets T0. The default profile includes a T[current_extruder] in "After layer change G-code".
Didn't expect a layer-change before printing the first layer, is this intentional behavior?

concerning temperature setpoints:
With RRF G-Gode flavor, SS generates extruder temperatures with `G10 P0 S230` -> this will exclusively set the Temperature for T0.
Other flavors like Klipper will generate `M104 S230` or `M109 S230` -> no reference which extruder should heat up!
Currently can´t test this on my machine, but I assume this might lead to problems, if the wrong tool is selected before temperatures are set.

btw Thanks for your work! this is actually the first time I had problems with your software.

supermerill, 2021-08-07:
> concerning temperature setpoints

For not-rrf, the M104 should heat up the current (or default?) extruder. I don't know if some more specific gcode like the G10 for rff exist.

> if the wrong tool is selected before temperatures are set.

It should change the heat to "park temp" before tool change and to "print temp" after. But if you're writing your own tool change, you may have to write it yourself.

johnmeacham, 2021-09-13:
I was able to work around the issue by moving my undocking the current tool to a macro so that superslicer does not see the toolchange in the custom gcode.

closed 2021-09-13 by johnmeacham