supermerill/SuperSlicer #1409: Del Key not working when editing settings

dkatzdev, 2021-07-13:
### Version
2.3.56.7

### Operating system type + version
Win 10 Pro x64

### Behavior
Not sure if this was present in prior versions, but the Del key seems ineffective when editing in textboxes for settings. For example, when editing a filament printing temperature, the backspace key will work fine, but the "Del" key doesn't do anything.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
N/A

supermerill, 2021-08-01:
@liftbag @LazaroFilm or other mac-user, can you test this error on macOS?
It seems that there is two delete key in the "long" version of a macOS keyboard (like the "magic").
Can you use both to delete a digit in "perimeters" and "sparse infill (density)" field? 
Can you test it in 2.3.56.7 (and latest nightly if something doesn't work)?

LazaroFilm, 2021-09-07:
> @liftbag @LazaroFilm or other mac-user, can you test this error on macOS?

Tried it with 2.3.57.0 and yes there is an behavior issue with the delete key. When you put the cursor at the beginning of the number and press `del` key, it deletes the number on the right of the cursor (normal), but it also moves the cursor all the way to the right of the number (not normal). 
In the video below, I first delete the number with `backspace`, then go to the beginning of the number and hit delete. then I have to it the left arrow to go back to the beginning of the number before hitting `del` again. 

https://user-images.githubusercontent.com/18351100/132383593-90f344e6-b5a1-4f57-b5ff-6d32c4025151.mov

readeral, 2022-08-21:
I just searched for this issue as the problem is still live - on the matter of the moving cursor to the end, this actually occurs after any interaction with the text box. Repeatable example: if I have the number 10 in an input, click to activate the input and move the cursor to the beginning, if I add a number 2, the result is '210' but the cursor has moved to the end of the text.