supermerill/SuperSlicer #1412: If supports are set to soluble, bridging isn't recognized any more.

codingcatgirl, 2021-07-13:
**Describe the bug**
If i set supports to "soluble", even if there are no supports, bridging seems to get deactivated and bridging areas get rendered just like bottom solid layers. is this a known problem?

**To Reproduce**
Steps to reproduce the behavior:
1. Set supports to soluble
2. Select an object with bridges
3. Observe the behavior
4. There might be other factors, i'm attaching the project file for reference.

[bridgetest.zip](https://github.com/supermerill/SuperSlicer/files/6811305/bridgetest.zip)

**Expected behavior**
Bridging areas should use bridging settings.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - 2.3.56.6

supermerill, 2021-07-20:
If "don't support bridge" is unchecked, it should create bridges.

codingcatgirl, 2021-07-20:
Thanks, that's great to know as a workaround, as i rarely use automated supports anyway.

supermerill, 2021-07-20:
I mena, it should create bridges, but it doesn't, it's a bug.

codingcatgirl, 2021-07-20:
Yeah, but it's still good to know that i can circumvent it like this.

supermerill, 2021-07-31:
has to do something when there is only enforcer or there is some blockers, as the flow & pattern may be not optimal.

codingcatgirl, 2021-07-31:
I'm a bit confused, as the example i uploaded has no enforcers or blockers at all?

supermerill, 2021-08-01:
The only reason of activating support without auto-generation is to use enforcers, or I forgot something?

codingcatgirl, 2021-08-01:
I have it always turned to supports enabled and automatic supports disabled so i can add enforcers if i want without having to change the settings, even though i don't use enforcers in most prints.

supermerill, 2021-08-02:
So, it need to recognize that and use bridging everywhere but over the eventual enforcer.

codingcatgirl, 2022-04-02:
This bug is back on the latest 2.4.* release. I can post a 3mf later. The workaround with "don't support bridges" does not fix it. Might be related to snug supports.

codingcatgirl, 2022-04-02:
Setting "No perimeters on bridge areas" to "keep only bridges" acts as a workaround.

codingcatgirl, 2022-04-02:
Never mind, as soon as supports are set to soluble, it's broken again.

3MF File: [test.zip](https://github.com/supermerill/SuperSlicer/files/8403048/test.zip)

Here's a screenshot that shows the issue, you can see the purple areas that should actually be bridges (blue).

[image]

codingcatgirl, 2022-04-02:
Removing all support enforcers or paint-on supports doesn't fix the problem either.

supermerill, 2022-04-02:
> This bug is back on the latest 2.4.* release. I can post a 3mf later. The workaround with "don't support bridges" does not fix it.

come from the merge. prusa has changed the next line a bit and I missed it.
will be fixed in sunday nightly.

codingcatgirl, 2022-04-25:
Even with the workaround, bridging perimeters aren't generated at all.

supermerill, 2022-04-26:
> However, checking "don't suport bridges" fixes it. Still, this should work with that setting disabled too, right?

I fixed the 'don't suport bridges', i didn't work on the feature request.