supermerill/SuperSlicer #1414: Fan speed settings for "internal" bridges

CageFox, 2021-07-14:
**Is your feature request related to a problem? Please describe.**
Current version have no ability to use different fan speed for "internal" bridges, it uses fan speed for normal bridges. This is deadly for plastics like ABS, where I still want to print normal bridges with serious fan speed, but do not want to use fan at all in the rest of the print

**Describe the solution you'd like**
Parameter in cooling section of filament - fan speed for internal bridges

**Describe how it would work**
Use this speed for fan while printing internal bridges

**Describe alternatives you've considered**
For now only alternative is to lower bridge speed as much as possible

[image]

gpread, 2021-09-07:
Just tried this as I also have scenarios, where I don't want any fan except for bridges and bridge fan speed for in internal bridge fill (which are much larger areas) causes problems to other subsequently printed features/layers (I have even seen it cause delamination/layer bonding problems). 

Neither the FAN speed view nor the g-code output appears to respect the new setting applied in Filament Settings->Cooling, the behaviour is just the same as before I changed the "infill bridges" from its initial -1 to 20%

[image]

[image]

[image]

It clearly does do something with the 20% setting requested since in it correctly describes the desired effect in the "Behaviour" summary at the bottom of the Filament Settings-Cooling Dialog

gpread, 2021-09-07:
[Bridge Infill Fan speed (#1414) not being respected.zip](https://github.com/supermerill/SuperSlicer/files/7124703/Bridge.Infill.Fan.speed.1414.not.being.respected.zip)

gpread, 2021-09-07:
2.3.57.0, Creality CR-10V3, Marlin 2.0.8.2 & Windows 10 - 20H2 (19402.1165)

CageFox, 2021-09-10:
Confirm: this settings not working. It keep 100% fan speed on internal bridge, even if I enter 20% in new setting
[image]

CageFox, 2021-09-10:
I see, the feature is disabled it the infill bridges fan speed equals or lower then default fan speed. This is counter intuitive, the same speed as default speed should wortk as well. And I even think there must be no dependency upon default speed - if infill perimeters speed specified - it whould work regardless of default speed
[image]

closed 2021-10-08 by supermerill