supermerill/SuperSlicer #1415: Short layer time when using supports

maziggy, 2021-07-15:
When using supports and settings 'max fan speed for layer print time below' fan speeds will increase for support layer but don't decrease after it was finished and the print continue with a 'standard' layer although the print time is far above the setting.

Version 2.3.56.7 on MacOS 11.2.3

[image]

[smuff-set-red.gcode.zip](https://github.com/supermerill/SuperSlicer/files/6823474/smuff-set-red.gcode.zip)

maziggy, 2021-07-16:
In the meantime I also tried to disable 'Run fan ath ethe default...' but b´nothing changed. This is also the case for the latest version 2.3.56.7.

supermerill, 2021-07-20:
I need your project file

maziggy, 2021-07-21:
Unfortunately I already deleted the file. Will try to reproduce and send you the file.