supermerill/SuperSlicer #1416: External perimiter still gets fan

mrvanes, 2021-07-15:
I set outer perimiter value to 1 (explanation says this effectively disables fan on external perimiters) but the fan speed picture clearly shows the fan will run at the default speed (10) for some of the external perimiters?

**To Reproduce**
Steps to reproduce the behavior:
1. Enable "Run the fan at default speed when possible"
2. Set default fan speed
3. Set external perimiter fan speed to 1
4. See error in Gcode preview 'fan speed'

**Project File**
The attached file is the Klipper3D pressure advance tuning tower as an example.
[square_tower.zip](https://github.com/supermerill/SuperSlicer/files/6823919/square_tower.zip)

**Expected behavior**
Even though I selected the fan to run at default speed when possible, I would expect the fan to be forced off on external perimiter when set to 1.

**Screenshots**
[image]

**Desktop**
 - OS: Linux
 - Version 2.3.56.7

neophyl, 2021-07-16:
Let me guess (as you haven't included a saved 3mf project file to help us debug the issue and the stl helps in no way to know what your settings actually are).  

My crystal ball is coming into focus, mmm the spirits are telling me to look in your Printer Settings.  Yes Printer Settings >General>Cooling Fan.  There is more, the spirits are also telling me that you need to check if you have Speedup time enabled.

I kid :)  Its Friday morning and I need coffee.  In all seriousness theres a fan speedup function that makes the fan kick in earlier so that its up to speed for upcoming moves, like bridging.  It was added by user request and I'm guess you have something like -1 seconds in there which means that while you may have perimeters fan turned off it kicks in for the other extrusions early.  
I see the same with first layer, first layer is off but the final bit of the printing is with the fan on.

mrvanes, 2021-07-16:
Ok, this is embarassing... and take care of the crystal ball, it's priceless.
(You were right)

closed 2021-07-16 by mrvanes