supermerill/SuperSlicer #1417: Brim Extrusion Width (Separated from Skirt)

InspiredMartin, 2021-07-15:
When trying to print a brim, I'm finding that the extrusion width is tight, causing a lot of warping on the brim portion of the first layer. It seems like this setting might be hardcoded as a small percentage of first layer extrusion width, to be able to make sure it adheres well to the part.

It would be really helpful to be able to dial this setting in on our own, so having a separate brim extrusion width setting would be helpful.

An alternative would just be to have the brim width follow the same setting as the skirt width.

supermerill, 2021-07-20:
It should use the first_layer_extrusion_width (if set, or the first_layer_extrusion_spacing). Same for the skirt width.
Do you really have a use-case where it needs to be different?

InspiredMartin, 2021-07-21:
I guess I can clarify, I don't have a use case where it needs to be different, it just doesn't seem like it's actually following: These brims are not at the same spacing as the rest of the layer. All of my brims have been coming out warped because filament lines are overlapping too much.
[image]

supermerill, 2021-07-21:
can you drop here the (zipped) project file where you encounter this behavior?