supermerill/SuperSlicer #1418: compilation error

Tinchus2009, 2021-07-15:
**Describe the bug**
Hi. Im try to compile superslicer follwing the "Noob instructions" from the wifi. When I try to run the compiled version (all steps finished with no errors_ in visual studio, I get this error in the console:

Slic3r.dll was not loaded, error code: 126

What step Im missing  / doing wrong?

supermerill, 2021-07-16:
I guess you're on windows
add libmpfr-4.dll and libgmp-10.dll next to your exe

Tinchus2009, 2021-07-17:
they are already there:

[image]

supermerill, 2021-07-20:
Weird. Try to add the visual c++ dll (v140), maybe your installation doesn't add them to the path.