supermerill/SuperSlicer #1420: Inner Layer Fan speed as Percentage/Offset of Auto Fan Speed

DontTakeItForGranted, 2021-07-16:
**Describe the solution you'd like**
The fan speed setting  for the external perimeters is static, which means you cant use the auto fan speed function for external layers which further means you get ether over or under cooling on external perimeters. Therefore i would suggest to control the external perimeter fan speed as usual with the auto fan speed function, but control the inner layer fan speed (any layer except external ,bridge and first) as percentage/offset of the auto fan speed, this way you have still adaptive cooling on external and inner layers, but depending on how you set it the inner layer fan would be off at longer layer times and just barely spinning on short layer times.
with this setting you could also just put a 0 in and your inner layers would have cooling off all the time but your external perimeter fan speed is still adaptive controlled. With this setting you have basically all doors open for different scenarios.

**Describe how it would work**
1. you would have an extra value setting box where you put in your percentage of the auto fan speed for inner layers 
what it would do:

Auto fan speed (as usual, adaptive controlled by layer time)*Inner layer speed as percentage of auto speed = Inner layer Fan Speed 
**Example_a:** 
73% * 50% = 37%  (no Arithmetic Operator given by the user)
 
An additional part of this feature would be if you put a minus before the value in the "percentage of ..." box it is recognised as inner layer fan speed offset from the value chosen by auto fan speed
 **Example_b:**
73% - 50% =  23% (minus is Given by the user)

73% stands representative for the fan speed value that was chosen by the auto fan speed feature based on the layer time
50% ; - 50% are representative for the value given by the user(“inner layer fan speed as percentage of auto speed)
37% ; 23% are representative for the actual inner layer fan speed

2. if the actual inner layer fan speed would fall below the default/min value it is kept at this value. but there should be an extra checkbox called "set inner layer fan to 0 if it falls below the default" with this box checked the fan is turned of if the inner layer fan speed falls below the default fan speed value.
Example:
(DEFAULT/MIN SPEED = 20%)
**BOX NOT CHECKED**
30% * 50% = 20% (kept at min/default)
30% - 50% = 20%
**BOX CHECKED**
30% *  50% = 0%  (set to 0 because below min/default)
30% - 50% = 0%

30% stands representative for the fan speed value that was chosen by the auto fan speed feature based on the layer time
50% ; - 50% are representative for the value given by the user(“inner layer fan speed as percentage of auto speed)
20% ; 0% are representative for the actual inner layer fan speed

This feature would be just amazing for many use cases, i hope my description isn´t to cryptic (my english isn`t that great), if you need further explanation i will be happy to provide it.

**HAVE A NICE DAY:)!**

DontTakeItForGranted, 2021-07-16:
This Feature would give you a way better control over how much less cooling you have on the inside of a model and therefore increase the layer adhesion while preserving the normal printing quality (both the inner and external layers are still adaptive controlled this way, the inner layer is just less cooled depending on how you set it)

supermerill, 2021-07-20:
You're right, i forgot to scale the fan speed override the same way the default one works.

DontTakeItForGranted, 2021-07-21:
do you consider implementing this feature just as scalability for the external layers, or for the inner layers as i proposed it?
(i am not sure how to interpret your comment)
This is important because when you just scale the external layer fan speed based on the total (auto ) fan speed you wouldn`t  be able to for example turn the inner layer (in this case it would be also the total/auto fan speed)  speed the to 0 and still have a changing fan speed for the external perimeters, because a percentage value of 0 is still just zero and a offset of 0 is still just  the offset, so the sped for the external perimeter would be still constant (in this scenario, the range of scaling would be also not that big for other scenarios). 
HAVE A NICE DAY :)!

supermerill, 2021-07-21:
I have to find a spare time when i'm fresh to re-read your proposal, I don't understand sentences like 
`I would suggest controlling the external perimeter fan speed as usual with the auto fan speed function, but control the inner layer fan speed of the auto fan speed` easily.

DontTakeItForGranted, 2021-07-25:
I did not mean to sound rude in my last comment, i am really thankful for your work and time.
Because my text was hard to understand i made two images with photo editing to show what i had in mind.

What i meant with auto fan speed was the fan speed the external perimeter has when you put a -1 in the box,
this is unclear to understand because of that i called it just "inner layer fan speed percentage/offset of external perimeter speed" this means when the external perimeter fan speed is adaptive controlled by layer time (-1 in its box) the inner layer fan 
speed is just the percentage/offset of that speed.

inner layers have 50% of the external perimeter speed 
example: 70%*50%=35% (35%=inner layer fan speed)
[image]

inner layers have a offset fan speed of -30 from the external perimeter fan speed
example: 70%-30=40%    (40%=inner layer fan speed)
[image]

Behavior of the " keep minimum inner layer fan speed at default fan speed" box,

1. box not checked: inner layer fan speed is set to zero if it would fall below default fan speed 

2. box checked:  inner layer fan speed is kept at default fan speed if possible.

NOTE: this box behaves exactly the same as the "run fan at default speed when possible" box but only for the inner layer fan speed,  this additional box would give you a independent control over your minimum fan speed for inner and external layers.

Lets say you checked the "run fan at default speed when possible" box and did not checked the "keep minimum inner layer fan speed at default fan speed" box, 
now your external perimeters fan speed could only go down to the default fan speed 
but your inner layer fan speed could go down to zero.

Have a Great Day:)

supermerill, 2021-07-26:
Thank you, i understand better.

You want to have higher fan speed for external periemter vs other extrusion type, but only if the layer time is too low. It's right?
What is your current problem that you want to solve with this?

I still don't understand what an "external layer" is, is it a top surface?

DontTakeItForGranted, 2021-07-26:
yes this was the idea of the added check box.

I basically just added this check box to let it unchecked (in real use),
i did this because i think it would be confusing if you have the  "run fan at default speed when possible" box checked 
but your inner layer fan speed would still go down to zero percent.

Another solution would be  just to rename the "run fan at default speed when possible" box to 
"run external layer fan at default speed when possible".
that way the inner layer fan speed could go to zero even if that box is checked, 
because the renamed box just says something about the fan speed behavior on external layers. (it is more clearly)

### but all in all this box is not that important 
the important enhancement/feature is really just the percentage/offset thing so please don`t spend to much 
time decrypting my unclear “explanations“.

With external layers i meant: external Perimeter and Top surface   
With inner layers i meant: non external perimeters (just perimeters), infill, solid infill, Gap fill
NOTE: bridges are still controlled as normal (100% fan speed) even if they are inside of a print.  

### Have a super nice day:)