supermerill/SuperSlicer #1421: Add SuperSlicer package manager package/PPA for continuous updates without the need to pull latest release

Meganium97, 2021-07-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I'd like to be able to update SuperSlicer without the need to go to the releases page every time and grab a new Appimage/tarball
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add a version of superslicer to many package managers through the use of something like a PPA (or something not too dissimilar)
**Describe how it would work**
Be as descriptive as your skill level allows you.
Add the Superslicer program to the repository lists (like apt, pacman, etc) and release it so that people can get continuous updates, while also getting the option to launch alongside other programs without the need to use the termina
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
An alternative could be the use of a PPA, like Cura's PPA.
**Additional context**
Add any other context or paint drawing about the feature request here.
None

Meganium97, 2021-07-27:
I am currently attempting to create a PPA using the official documentation, but am currently stuck on the building phase.
Here is the code I first attempted to use:
https://github.com/linuxmint/hypnotix/issues/125#issuecomment-813092266
However, it always returned with the error: `cannot find readable debian/changelog anywhere!`
I also tried to use cmake to at least see if I could get anywhere. Here is the error I ran into:

-- Could NOT find DBUS (missing: DBUS_INCLUDE_DIRS DBUS_LIBRARIES) 
CMake Error at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time) (Required is at least version
  "1.64.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:2177 (find_package_handle_standard_args)
  CMakeLists.txt:291 (find_package)

In the CMakeError.log file, it complained of:

/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x5e): undefined reference to `pthread_cancel'
/usr/bin/ld: src.c:(.text+0x6f): undefined reference to `pthread_join'

I am very confused as of what exactly I am doing wrong here. I will update when I eventually do.

supermerill, 2021-07-28:
> am currently stuck on the building phase.

DId you tried the `BuildLinux.sh` script?

Meganium97, 2021-07-28:
Yes. It did seem to work but did not in the end, instead returning the same cmakeerror.log error I meantioned earlier.
However, through the use of makedeb and a community-oriented PKGCONFIG file, I learned that I had packages to install, as I got this error:

The following build dependencies are missing: boost (>= 1.73.0) but it is not installable, cgal but it is not installable, eigen but it is not installable, glew but it is not installable, libigl but it is not installable, nlopt but it is not installable, openvdb but it is not installable, qhull (>= 2020.2-4) but it is not installable, wxgtk2-dev-opt but it is not installable, wxgtk3-dev-opt but it is not installable

I installed all packages needed yesterday, but ran out of time to retry the build process. I will update the issue when I have completed it.

Meganium97, 2021-07-28:
This time I got further, but TBB was needed to be built. Currently building.

Meganium97, 2021-07-28:
Now that the cmake stage has completed, my move into the make stage is now littered with errors. Here's the error that crashed it:

/home/ian/SuperSlicer/src/admesh/stlinit.cpp: In function ‘bool stl_read(stl_file*, FILE*, int, bool)’:
/home/ian/SuperSlicer/src/admesh/stlinit.cpp:204:52: warning: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘stl_normal’ {aka ‘class Eigen::Matrix<float, 3, 1, 2>’} with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
  204 |       memset(&facet.normal, 0, sizeof(facet.normal));
      |                                                    ^
In file included from /usr/include/eigen3/Eigen/Core:463,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /home/ian/SuperSlicer/src/admesh/stl.h:31,
                 from /home/ian/SuperSlicer/src/admesh/stlinit.cpp:33:
/usr/include/eigen3/Eigen/src/Core/Matrix.h:178:7: note: ‘stl_normal’ {aka ‘class Eigen::Matrix<float, 3, 1, 2>’} declared here
  178 | class Matrix
      |       ^~~~~~
/home/ian/SuperSlicer/src/admesh/util.cpp: In function ‘void stl_translate(stl_file*, float, float, float)’:
/home/ian/SuperSlicer/src/admesh/util.cpp:73:20: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
   73 |  for (int i = 0; i < stl->stats.number_of_facets; ++ i)
      |                  ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/util.cpp: In function ‘void stl_translate_relative(stl_file*, float, float, float)’:
/home/ian/SuperSlicer/src/admesh/util.cpp:84:20: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
   84 |  for (int i = 0; i < stl->stats.number_of_facets; ++ i)
      |                  ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/util.cpp: In function ‘void stl_scale_versor(stl_file*, const stl_vertex&)’:
/home/ian/SuperSlicer/src/admesh/util.cpp:103:20: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  103 |  for (int i = 0; i < stl->stats.number_of_facets; ++ i)
      |                  ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/util.cpp: In function ‘void stl_repair(stl_file*, bool, bool, bool, float, bool, float, bool, int, bool, bool, bool, bool, bool, bool)’:
/home/ian/SuperSlicer/src/admesh/util.cpp:327:6: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  327 |      if (! tolerance_flag)
      |      ^~
/home/ian/SuperSlicer/src/admesh/util.cpp:329:7: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  329 |       if (! increment_flag)
      |       ^~
/home/ian/SuperSlicer/src/admesh/util.cpp:333:41: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  333 |  if (stl->stats.connected_facets_3_edge < stl->stats.number_of_facets) {
      |      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/util.cpp:336:46: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  336 |       if (stl->stats.connected_facets_3_edge < stl->stats.number_of_facets) {
      |           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/util.cpp:354:42: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  354 |   if (stl->stats.connected_facets_3_edge <  stl->stats.number_of_facets) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/util.cpp:363:42: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  363 |   if (stl->stats.connected_facets_3_edge <  stl->stats.number_of_facets) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/stlinit.cpp:169:10: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  169 |    fscanf(fp, " endsolid%*[^\n]\n");
      |    ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/stlinit.cpp:170:10: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  170 |    fscanf(fp, " solid%*[^\n]\n");  // name might contain spaces so %*s doesn't work and it also can be empty (just "solid")
      |    ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/stlinit.cpp:185:9: warning: ignoring return value of ‘char* fgets(char*, int, FILE*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  185 |    fgets(buf, 2047, fp);
      |    ~~~~~^~~~~~~~~~~~~~~
/home/ian/SuperSlicer/src/admesh/stlinit.cpp:189:10: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  189 |    fscanf(fp, " ");
      |    ~~~~~~^~~~~~~~~
/home/ian/SuperSlicer/src/admesh/stlinit.cpp:190:9: warning: ignoring return value of ‘char* fgets(char*, int, FILE*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  190 |    fgets(buf, 2047, fp);
      |    ~~~~~^~~~~~~~~~~~~~~

It fails after this, declaring "Error2 2". Stuck.

Meganium97, 2021-07-28:
I found the error that therminated it now. Still stuck though.
`igl/copyleft/cgal/mesh_boolean.h: No such file or directory`

Meganium97, 2021-07-29:
I figured out all the rest of the issues and build the project. The only problem I have now is exactly this one > https://github.com/supermerill/SuperSlicer/issues/1093

Meganium97, 2021-07-31:
Ubuntu 21.04. I think its a problem with building with the latest wxwidgets, but the only solution that I've found is to build with the glew libraries provided with prusaslicer. That being said, this isn't prusaslicer, and even if it was, I don't know how to do that.

supermerill, 2021-07-31:
Why building with the latest wxwidgets? the deps should build the supported one.

Meganium97, 2021-08-01:
I use the wxwidgets to build different things. I guess I can just install the one from the deps

supermerill, 2021-08-01:
you don't have to "install" it, if you link it statically like I do.

Meganium97, 2021-08-02:
The cript runs into a bit of a problem because it cant find wxwidgets

supermerill, 2021-08-02:
mmm, re-reading the linux script, it contains macos thingy, wondering why...
whatever

line 64 should download & build wxwidget
line 93 should build SuSi with the `-DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"` giving the directory where the compiled version of wxwidget is. Maybe that's where your problem is, it doesn't link to the correct folder.

Meganium97, 2021-08-04:
I see these following lines in the file provided, it might be the source of the problems:

- name: clean deps
      working-directory: ./deps/build
      run: rm -rf dep_*
    - name: mkdir build
      run: mkdir build

However, not including these instructions make no difference.

supermerill, 2021-08-05:
the deps install their useful files in the `destdir` directory. Removing the `deps_...` ones allow to save a huge amount of hard disk space to allow the next `make` to succeed.

Meganium97, 2021-08-05:
I got a decently long error which ended in

```collect2: error: ld returned 1 exit status
make[3]: *** [src/CMakeFiles/Slic3r.dir/build.make:173: src/superslicer] Error 1
make[2]: *** [CMakeFiles/Makefile2:594: src/CMakeFiles/Slic3r.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:601: src/CMakeFiles/Slic3r.dir/rule] Error 2
make: *** [Makefile:231: Slic3r] Error 2```

After running the make step. The error is laced with "undefined reference" errors.

Meganium97, 2021-08-29:
The last problem I have is with dependencies and ppa not being able to connect as a part of its build process. Currently I need the IGL library and the cereal library only available through GitHub. Is there any chance you could include the GitHub libraries in the repo itself?

supermerill, 2021-08-30:
> ppa not being able to connect to the internet

If you can access the SuSi source code, you should have an internet access?
If you don't have the internet access afterwards, you can't download them before?

> Currently I need the IGL library and the cereal library only available through GitHub.

I put the boost library into a github release page (because the official repo has a download limit). You don't have a problem with this one?

Meganium97, 2022-10-28:
Just updating the feed after a while - I've completely switched off of ubuntu, there's not much reason to keep this open anymore except the hope that this will be added. Given its been half a year, I'm pretty confident that if anyone wanted to help they would've. For the future, if anyone wants to do this, go ahead and reopen this issue, but I'll be closing this issue for the time being.

closed 2022-10-28 by Meganium97

pinkpawkitty, 2023-02-04:
@Meganium97 @supermerill I'd like to give this a shot (no promises though), but I can't find the work that was done so far?

Meganium97, 2023-02-04:
Give me until later today, I will send everything then.

pinkpawkitty, 2023-02-18:
@Meganium97 *friendly reminder* :)

Meganium97, 2023-02-18:
Sorry.

PPA is right here.

https://launchpad.net/~meganium97/+archive/ubuntu/superslicer