supermerill/SuperSlicer #1422: Impossible to setup a first layer speed higher than print speed

tirasec, 2021-07-17:
**Describe the bug**
When the first layer print speed (first_layer_speed) is set higher than the print speeds, the first layer speed is decreased to the print move speeds.

**To Reproduce**
Steps to reproduce the behavior:
- Set first layer speed  (default and Infill) higher than the print speeds 
- Generate the GCode
- The first layer speed is adjusted to the Print moves speed.

We need to speed up the first layer when printing PEEK filament on PEEK bed, in order to avoid to damage the Bed

Note : It works on Prusa Slicer 2.3.0, we can setup any First layer speed

**Expected behavior**
The first layer speed should be setup to the First layer speed value of the dailog box

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version : 2.3.56.7

supermerill, 2021-07-20:
That was a failsafe i added long ago because i didn't see why someone may want to have a bigger first layer speed.

Also, it allows  to have a slow perimeter speed, a medium first layer speed and a fast (internal) perimeter speed, and to use the slow external perimeter speed, instead to override it.

I don't see how to allow the two behavior... maybe a checkbox...

tirasec, 2021-08-10:
A check box : "Force fisrt layer to this speed"(or something like that) should be nice...

closed 2021-10-08 by supermerill