supermerill/SuperSlicer #1423: Hole size is smaller at the bottom layers

laptop-laptop, 2021-07-17:
**Describe the bug**

Printed hole size is smaller than the actual CAD model. For example, I have a block with some holes passing through from the top to the bottom. In the model the size is 9.55mm in diameter but in the actual printed one, the hole size at the bottom layer is 8.9mm in diameter. 0.65mm different. Top layers hole size has no problem. Same thing for smaller 4mm holes.

neophyl, 2021-07-18:
Thats probably down to the 'squish' needed to get a proper sticking first layer.  Its what the 'elephants foot' setting is for.  It moves all the external perimeters in on the first layer to counteract this.
Also as a general observation nearly all printed holes shrink slightly in practice so you are doing well if your holes on non first layers are an ok size.  Most people adjust in their design software to make them slightly bigger, like 3.1mm for a 3mm hole for example.  As it differs from printer to printer its hard to pin down an absolute number.

Oh and Super Slicer also has hole compensation settings in case you hadn't spotted them.

supermerill, 2021-07-20:
like neophyl said, set your first layer xy compenstation to something like -0.325

laptop-laptop, 2021-07-21:
Thank you. I am trying now. What value shall I try for XYcompensation->Holes?

supermerill, 2021-07-21:
I don't know. depends on your measurement.

laptop-laptop, 2021-07-21:
I tried -0.325 but for the supposed to be 9.55mm hole, the printed one is about 8.8mm. For 4.1mm holes, some of them is about 2.7mm while others are about 3.2mm.

Actually, by increasing the absolute amplitude of first layer xy compensation, are holes supposed to get larger or smaller?
The other days I did some tests:

At default value of -0.2, hole size 8.9mm. 
At -0.25, it was 8.8mm. 
At -0.3,  it was 9.25mm.
At -0.325 (today's test), it was 9.55mm 
At -0.4,  it was 8.84. 
At 0.5, it was 8.63mm.

About whether or not the build plate is correctly levelled, I did first layer calibration and the surface looked and felt smooth. Is that what you mean about correctly levelled?

supermerill, 2021-07-22:
> Is that what you mean about correctly levelled?

yes.

For your problem, i can't help much more with what you give us.
Post your project file and a picture of the printed part.

laptop-laptop, 2021-07-23:
For the hole compensation settings, how many layers from the bottom does it has an effect on? Is it possible for users to decide the number of layers to be affected by it?

supermerill, 2021-07-25:
> For the hole compensation settings, how many layers from the bottom does it has an effect on?

Layers where there is a detected hole (almost convex and round),

> Is it possible for users to decide the number of layers to be affected by it?

no

closed 2021-09-06 by supermerill