supermerill/SuperSlicer #1424: Where is elephant foot compensation in SuperSlicer 2.2.53?

laptop-laptop, 2021-07-17:
**Is your feature request related to a problem? Please describe.**

Hi, there is such feature in PrusaSlicer but I cannot find it in SuperSlicer 2.2.53. Could you please tell  me where it is? Perhaps under different name?

neophyl, 2021-07-18:
Print Settings>Slicing.  Modifying Slices.  Of course you need to be in advanced or expert mode for that section to appear.
Its called x/y compensation:first layer

supermerill, 2021-07-20:
You have to insert a negative value to have the same behavior than a positive one from prusaslicer.

laptop-laptop, 2021-07-21:
Thank you. Besides the sign, is elephant foot compensation in prusaslicer exactly the same as XY compensation-First layer in SuperSlicer?

supermerill, 2021-07-21:
yes

closed 2021-08-04 by laptop-laptop